bemasher/rtlamr #118: Itron 50w

gomer23, 2019-06-15:
Hello, I set up rtlamr yesterday and have been able to pull SCM from my gas and electric meter. But the itron 50w that is on my water meter doesn't seem to be showing up. I see that meter is on the supported list. I have the nooelec USB device from Amazon for reading the data and using a RPi3. Is there a different frequency or setting I need to change to pick up the messages from the 50w?

bemasher, 2019-06-16:
Battery-operated meters like the 50W are frequently operated in wake-up mode, which means they don't transmit unless woken up by the utility's transmitter. You might also check to see if it is transmitting SCM+.

gomer23, 2019-06-16:
That makes sense and I did check SCM+. It looks like newer models are doing SCM+ but it's sounding like it's in wakeup mode. Time to look at other options I suppose.

closed 2019-08-03 by bemasher

greyltc, 2024-05-04:
I have this meter too. Any idea how to trigger the wakeup?