bemasher/rtlamr #122: SK9AMI-4 Support

impala454, 2019-09-02:
I tried and failed to link this ancient disqus thread for this topic, suffice it to say I'm still on a quest to read my electric meter, an Itron C2S0D, FCC ID: SK9AMI-4.  I noticed this FCC ID is currently in the `meters.csv` file so wasn't sure if other folks already read this meter succesfully.  I currently run rtlamr on a NooElec NESDR Mini 2 with successful reads of both of my Itron gas and water meters.

[image]

Page 10 of the [FCC test report](https://fccid.io/SK9AMI-4/Test-Report/Test-Report-1229203) shows frequencies of 914.75 MHz and 927.75 MHz.  Could I use those frequencies to glean more information on how to read this meter?

I'm particularly ignorant of signals processing but know my way around software development so don't mind doing some coding to test anything you think may help.

bemasher, 2019-09-02:
Your meter's labeling suggests that it is using OpenWay (a ZigBee based protocol) rather than ERT. Unfortunately, rtlamr does not support OpenWay.

closed 2019-09-02 by bemasher

impala454, 2019-09-02:
Good to know, thanks.  I appreciate the quick reply.