bemasher/rtlamr #123: armv7 compile failure

kevinelliott, 2019-09-06:
```
 > [linux/arm/v7 builder 2/2] RUN go get github.com/bemasher/rtlamr && ls -l /root/go/bin:                                                
#11 14.64 # github.com/bemasher/rtlamr                                                                                                    
#11 14.64 runtime: pointer 0xa00020 to unused region of span span.base()=0x7e6000 span.limit=0x8063a0 span.state=1                        
#11 14.64 runtime: found in object at *(0xfffc5008+0x7098)                                                                                
#11 14.64 object=0xfffc5008 s=nil                                                                                                         
#11 14.64 fatal error: found bad pointer in Go heap (incorrect use of unsafe or cgo?)                                                     
#11 14.64 
#11 14.64 runtime stack:
#11 14.64 runtime.throw(0xffdefa18, 0x3e)
#11 14.64       /usr/lib/go/src/runtime/panic.go:608 +0x60 fp=0x435ee0 sp=0x435ecc pc=0xffc35720
#11 14.64 runtime.findObject(0xa00020, 0xfffc5008, 0x7098, 0x0, 0x0, 0x0)
#11 14.64       /usr/lib/go/src/runtime/mbitmap.go:399 +0x358 fp=0x435f04 sp=0x435ee0 pc=0xffc1aa40
#11 14.64 runtime.scanblock(0xfffc5008, 0x40000, 0xff302c00, 0x41c960)
#11 14.64       /usr/lib/go/src/runtime/mgcmark.go:1048 +0x98 fp=0x435f2c sp=0x435f04 pc=0xffc27a48
#11 14.64 runtime.markrootBlock(0xfffc5008, 0x6320, 0xff302c00, 0x41c960, 0x0)
#11 14.64       /usr/lib/go/src/runtime/mgcmark.go:294 +0x5c fp=0x435f40 sp=0x435f2c pc=0xffc25804
#11 14.64 runtime.markroot(0x41c960, 0x2)
#11 14.64       /usr/lib/go/src/runtime/mgcmark.go:195 +0x348 fp=0x435f84 sp=0x435f40 pc=0xffc25708
#11 14.64 runtime.gcDrain(0x41c960, 0xd)
#11 14.64       /usr/lib/go/src/runtime/mgcmark.go:882 +0x118 fp=0x435fcc sp=0x435f84 pc=0xffc271a8
#11 14.64 runtime.gcBgMarkWorker.func2()
#11 14.64       /usr/lib/go/src/runtime/mgc.go:1862 +0x188 fp=0x435fe8 sp=0x435fcc pc=0xffc5f740
#11 14.64 runtime.systemstack(0x42a3c0)
#11 14.64       /usr/lib/go/src/runtime/asm_arm.s:354 +0x90 fp=0x435fec sp=0x435fe8 pc=0xffc61738
#11 14.64 runtime.mstart()
#11 14.64       /usr/lib/go/src/runtime/proc.go:1229 fp=0x435fec sp=0x435fec pc=0xffc3a2f4
#11 14.64 
#11 14.64 goroutine 18 [GC worker (idle)]:
#11 14.64 runtime.systemstack_switch()
#11 14.64       /usr/lib/go/src/runtime/asm_arm.s:298 +0x4 fp=0x426798 sp=0x426794 pc=0xffc6169c
#11 14.64 runtime.gcBgMarkWorker(0x41c000)
#11 14.64       /usr/lib/go/src/runtime/mgc.go:1826 +0x1b8 fp=0x4267e4 sp=0x426798 pc=0xffc23020
#11 14.64 runtime.goexit()
#11 14.64       /usr/lib/go/src/runtime/asm_arm.s:867 +0x4 fp=0x4267e4 sp=0x4267e4 pc=0xffc63458
#11 14.64 created by runtime.gcBgMarkStartWorkers
#11 14.64       /usr/lib/go/src/runtime/mgc.go:1720 +0x74
#11 14.64 
#11 14.64 goroutine 1 [runnable]:
#11 14.64 runtime.semrelease(0xfffdae1c)
#11 14.64       /usr/lib/go/src/runtime/sema.go:155 +0x30 fp=0x6d2f84 sp=0x6d2f84 pc=0xffc49cf4
#11 14.64 runtime.gcStart(0x0, 0x1, 0x0, 0x0, 0x0)
#11 14.64       /usr/lib/go/src/runtime/mgc.go:1387 +0x37c fp=0x6d2fb0 sp=0x6d2f84 pc=0xffc21618
#11 14.64 runtime.mallocgc(0x4000, 0xffeb1e10, 0x83b201, 0x1f)
#11 14.64       /usr/lib/go/src/runtime/malloc.go:996 +0x3f4 fp=0x6d3014 sp=0x6d2fb0 pc=0xffc1325c
#11 14.64 runtime.growslice(0xffeb1e10, 0x832000, 0xc00, 0xc00, 0xc01, 0x832000, 0x940, 0xc00)
#11 14.64       /usr/lib/go/src/runtime/slice.go:204 +0x128 fp=0x6d3044 sp=0x6d3014 pc=0xffc4dab8
#11 14.64 cmd/link/internal/objfile.(*objReader).readRef(0x6d322c)
#11 14.64       /usr/lib/go/src/cmd/link/internal/objfile/objfile.go:391 +0x514 fp=0x6d30cc sp=0x6d3044 pc=0xffd1962c
#11 14.64 cmd/link/internal/objfile.(*objReader).loadObjFile(0x6d322c)
#11 14.64       /usr/lib/go/src/cmd/link/internal/objfile/objfile.go:109 +0x284 fp=0x6d31c4 sp=0x6d30cc pc=0xffd16b24
#11 14.64 cmd/link/internal/objfile.Load(0xfffc77b0, 0x46e0f0, 0x452388, 0x46c3f0, 0x27fcee, 0x0, 0x74bef0, 0x2b)
#11 14.64       /usr/lib/go/src/cmd/link/internal/objfile/objfile.go:69 +0x140 fp=0x6d32b8 sp=0x6d31c4 pc=0xffd16694
#11 14.64 cmd/link/internal/ld.ldobj(0x458090, 0x452388, 0x46c3f0, 0x27fd91, 0x0, 0x74bef0, 0x2b, 0x6cb6af, 0x23, 0x0)
#11 14.64       /usr/lib/go/src/cmd/link/internal/ld/lib.go:1550 +0xa64 fp=0x6d343c sp=0x6d32b8 pc=0xffd73ce8
#11 14.64 cmd/link/internal/ld.loadobjfile(0x458090, 0x46c3f0)
#11 14.64       /usr/lib/go/src/cmd/link/internal/ld/lib.go:846 +0x428 fp=0x6d3524 sp=0x6d343c pc=0xffd6e044
#11 14.64 cmd/link/internal/ld.(*Link).loadlib(0x458090)
#11 14.64       /usr/lib/go/src/cmd/link/internal/ld/lib.go:361 +0x174 fp=0x6d3674 sp=0x6d3524 pc=0xffd6a960
#11 14.64 cmd/link/internal/ld.Main(0xfffc77b0, 0x4, 0x8, 0x1, 0xd, 0xe, 0xffde6375, 0x17, 0xffde6b4a, 0x18, ...)
#11 14.64       /usr/lib/go/src/cmd/link/internal/ld/main.go:207 +0xb04 fp=0x6d3738 sp=0x6d3674 pc=0xffd83ee4
#11 14.64 main.main()
#11 14.64       /usr/lib/go/src/cmd/link/main.go:65 +0x168 fp=0x6d3fc4 sp=0x6d3738 pc=0xffdde008
#11 14.64 runtime.main()
#11 14.64       /usr/lib/go/src/runtime/proc.go:201 +0x260 fp=0x6d3fe4 sp=0x6d3fc4 pc=0xffc37138
#11 14.64 runtime.goexit()
#11 14.64       /usr/lib/go/src/runtime/asm_arm.s:867 +0x4 fp=0x6d3fe4 sp=0x6d3fe4 pc=0xffc63458
#11 14.64 
#11 14.64 goroutine 2 [force gc (idle)]:
#11 14.64 runtime.gopark(0xffeb786c, 0xfffda5d0, 0x1410, 0x1)
#11 14.64       /usr/lib/go/src/runtime/proc.go:302 +0xe4 fp=0x426fc0 sp=0x426fac pc=0xffc3761c
#11 14.64 runtime.goparkunlock(0xfffda5d0, 0x1410, 0x1)
#11 14.64       /usr/lib/go/src/runtime/proc.go:308 +0x40 fp=0x426fd4 sp=0x426fc0 pc=0xffc376c0
#11 14.64 runtime.forcegchelper()
#11 14.64       /usr/lib/go/src/runtime/proc.go:251 +0xc8 fp=0x426fec sp=0x426fd4 pc=0xffc37478
#11 14.64 runtime.goexit()
#11 14.64       /usr/lib/go/src/runtime/asm_arm.s:867 +0x4 fp=0x426fec sp=0x426fec pc=0xffc63458
#11 14.64 created by runtime.init.3
#11 14.64       /usr/lib/go/src/runtime/proc.go:240 +0x28
#11 14.64 
#11 14.64 goroutine 3 [GC sweep wait]:
#11 14.64 runtime.gopark(0xffeb786c, 0xfffda6a8, 0xffc2140c, 0x1)
#11 14.64       /usr/lib/go/src/runtime/proc.go:302 +0xe4 fp=0x4277c0 sp=0x4277ac pc=0xffc3761c
#11 14.64 runtime.goparkunlock(0xfffda6a8, 0xffe1140c, 0x1)
#11 14.64       /usr/lib/go/src/runtime/proc.go:308 +0x40 fp=0x4277d4 sp=0x4277c0 pc=0xffc376c0
#11 14.64 runtime.bgsweep(0x41a040)
#11 14.64       /usr/lib/go/src/runtime/mgcsweep.go:52 +0x90 fp=0x4277e4 sp=0x4277d4 pc=0xffc28dd0
#11 14.64 runtime.goexit()
#11 14.64       /usr/lib/go/src/runtime/asm_arm.s:867 +0x4 fp=0x4277e4 sp=0x4277e4 pc=0xffc63458
#11 14.64 created by runtime.gcenable
#11 14.64       /usr/lib/go/src/runtime/mgc.go:216 +0x48
#11 14.64 
#11 14.64 goroutine 17 [finalizer wait]:
#11 14.64 runtime.gopark(0xffeb786c, 0xfffecc4c, 0x140f, 0x1)
#11 14.64       /usr/lib/go/src/runtime/proc.go:302 +0xe4 fp=0x42278c sp=0x422778 pc=0xffc3761c
#11 14.64 runtime.goparkunlock(0xfffecc4c, 0x140f, 0x1)
#11 14.64       /usr/lib/go/src/runtime/proc.go:308 +0x40 fp=0x4227a0 sp=0x42278c pc=0xffc376c0
#11 14.64 runtime.runfinq()
#11 14.64       /usr/lib/go/src/runtime/mfinal.go:175 +0xac fp=0x4227ec sp=0x4227a0 pc=0xffc1e07c
#11 14.64 runtime.goexit()
#11 14.64       /usr/lib/go/src/runtime/asm_arm.s:867 +0x4 fp=0x4227ec sp=0x4227ec pc=0xffc63458
#11 14.64 created by runtime.createfing
#11 14.64       /usr/lib/go/src/runtime/mfinal.go:156 +0x6c
#11 14.64 
#11 14.64 goroutine 19 [GC worker (idle)]:
#11 14.64 runtime.gopark(0xffeb77d0, 0x83c430, 0x1417, 0x0)
#11 14.64       /usr/lib/go/src/runtime/proc.go:302 +0xe4 fp=0x422f98 sp=0x422f84 pc=0xffc3761c
#11 14.64 runtime.gcBgMarkWorker(0x41d300)
#11 14.64       /usr/lib/go/src/runtime/mgc.go:1772 +0xe4 fp=0x422fe4 sp=0x422f98 pc=0xffc22f4c
#11 14.64 runtime.goexit()
#11 14.64       /usr/lib/go/src/runtime/asm_arm.s:867 +0x4 fp=0x422fe4 sp=0x422fe4 pc=0xffc63458
#11 14.64 created by runtime.gcBgMarkStartWorkers
#11 14.64       /usr/lib/go/src/runtime/mgc.go:1720 +0x74
```

bemasher, 2019-09-07:
Can you give some details about the processor, system and os you're building this on?

kevinelliott, 2019-09-07:
docker buildx, which does arm/v7, arm64, and amd64. Not problems with the amd64.

bemasher, 2019-09-07:
Can you provide the output for each of the following commands?

 * `uname -a`
 * `cat /etc/os-release`
 * `go version`
 * `go env`
 * `ldd rtlamr`

kevinelliott, 2019-09-11:
#31 [linux/arm64 builder 2/6] RUN uname -a
#31 0.306 Linux buildkitsandbox 4.9.184-linuxkit #1 SMP Tue Jul 2 22:58:16 UTC 2019 aarch64 Linux
```

```
#32 [linux/arm64 builder 3/6] RUN cat /etc/os-release
#32 0.218 NAME="Alpine Linux"
#32 0.218 ID=alpine
#32 0.218 VERSION_ID=3.9.4
#32 0.218 PRETTY_NAME="Alpine Linux v3.9"
#32 0.218 HOME_URL="https://alpinelinux.org/"
#32 0.218 BUG_REPORT_URL="https://bugs.alpinelinux.org/"
```

```
#33 [linux/arm64 builder 4/6] RUN go version
#33 0.493 go version go1.11.5 linux/arm64
```

```
#34 [linux/arm64 builder 5/6] RUN go env
#34 1.179 GOARCH="arm64"
#34 1.181 GOBIN=""
#34 1.181 GOCACHE="/root/.cache/go-build"
#34 1.181 GOEXE=""
#34 1.181 GOFLAGS=""
#34 1.182 GOHOSTARCH="arm64"
#34 1.183 GOHOSTOS="linux"
#34 1.183 GOOS="linux"
#34 1.183 GOPATH="/root/go"
#34 1.183 GOPROXY=""
#34 1.184 GORACE=""
#34 1.185 GOROOT="/usr/lib/go"
#34 1.185 GOTMPDIR=""
#34 1.185 GOTOOLDIR="/usr/lib/go/pkg/tool/linux_arm64"
#34 1.185 GCCGO="gccgo"
#34 1.186 CC="gcc"
#34 1.186 CXX="g++"
#34 1.187 CGO_ENABLED="1"
#34 1.188 GOMOD=""
#34 1.188 CGO_CFLAGS="-g -O2"
#34 1.188 CGO_CPPFLAGS=""
#34 1.188 CGO_CXXFLAGS="-g -O2"
#34 1.189 CGO_FFLAGS="-g -O2"
#34 1.189 CGO_LDFLAGS="-g -O2"
#34 1.189 PKG_CONFIG="pkg-config"
#34 1.190 GOGCCFLAGS="-fPIC -pthread -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build734129006=/tmp/go-build -gno-record-gcc-switches"
```

bemasher, 2019-09-15:
At the moment I'm unable to reproduce your error. I've attempted to do so using alpine linux on an RPi3 with no luck.

In any case, this appears to be an issue with your build environment rather than an issue with rtlamr. Since go will happily cross-compile from and to any architecture and OS, I would recommend building rtlamr on a known-working build environment for the target system you're interested in as follows starting in the rtlamr source directory:

```bash
GOOS=linux GOARCH=arm64 go build -v .
```

closed 2019-09-15 by bemasher