bemasher/rtlamr #126: MK7C Support

jLynx, 2019-10-09:
Hi all

So I am very new to this all. I just moved into a new house and noticed it has a MK7C smart meter.
The spec sheet https://www.meteringdynamics.com.au/wp-content/uploads/2017/06/8151-EDMI-Mk7C-WC-Product-Fact-Sheet.pdf states that it has a frequency range of 45-65Hz.

From the small ammount of research I have done already, it seems to post data every 30 minutes.
I am wondering what I would need to do in order to try and get reading from this.
(I have a RTL-SDR, HackRF and a YardStick One at my desposal to try and work this out)

mash55, 2019-10-09:
The RTL-SDR has a frequency range of 500kHz to 1.7GHz. Not sure about the HackRF. However, I doubt that it sends data between 45-65Hz.

bemasher, 2019-10-10:
I think the spec referring to 45-65 Hz is the useful frequency range of the alternating current it can measure, not the frequency that it transmits at (if it does at all).

jLynx, 2019-10-10:
Digging into this deeper, it seems like it transmits on 3G which means sadly, this wont be able to work. If anyone has anymore informaion please feel free to post it here.

closed 2019-10-10 by jLynx

tbertels, 2023-01-01:
The EDMI Mk7C can be connected to different 2G/3G/4G/LTE modems through RS-232.
https://www.edmi-meters.com/wordpress/wp-content/uploads/2018/03/Mk7C-Factsheet-English.pdf

I've got one connected to an EDMI EWM100 (Wavecom GSM chipset version).
[The datasheet](https://scribd.com/doc/308343265/Ewm100-Integra) says GSM quadband (850/900/1800/1900), 2G & 2.5G support.

There's also a Sierra chipset version.
[The datasheet](https://memoco.eu/media/wp-content/uploads/2011/06/EWM100-GSM-GPRS.pdf) says GSM quadband (850/900/1800/1900) too, GSM + GPRS Class 10 support.

tbertels, 2023-07-09:
Based on the [manual](https://www.camax.co.uk/downloads/EDMI-Atlas-Hardware-Reference-Manual-Revision-M.pdf#page=88) page 88 and followings:
  
[image]

Starting at [page 94](https://www.camax.co.uk/downloads/EDMI-Atlas-Hardware-Reference-Manual-Revision-M.pdf#page=94), the different communication options are described.

If you don't have access to the connections, you can also use a light detector to count the pulses, see https://github.com/colwilliamsnz/Power-Meter-for-ESPHome