bemasher/rtlamr #129: Not sure how to read the output

robman501a, 2019-11-07:
Hello,
This isn't an issue at all but we just got a new Neptune R900M water meter and it is supported by rtlamr but when I run the program, I don't know what I am looking at. I am able to get packets but I can't make heads or tails with it.
Is there a guide somewhere so I know how to read the data? 
This is one of the packets it was able to decode.
{Time:2019-11-06T20:58:17.427 R900:{ID:xxxxxxxxx  Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption:    9931 Unkn3:0x00 Leak: 5 LeakNow:2}}

jasonyork, 2019-11-07:
I think this is what you're looking for: 
https://github.com/bemasher/rtlamr/wiki/Protocol#r900-consumption-message

robman501a, 2019-11-08:
Just checked it over and no, sorry that doesn't help.
I've located our meter with the SN that's on the sticker of our Neptune meter.
Unkn1:0xA3 ??
NoUse:32 - 32 days of not using water?
BackFlow:0  ??
Consumption:     582  What unit of measure is this? 
Unkn3:0x00  ??
Leak: 0    - okay, so no leaks. That's good.
LeakNow:0  - same with above.

robman501a, 2019-11-15:
Hi, still no clue how to read the output. So, what would this mean??? 
Consumption:    2506
Gallons? Litres?

kb1ca, 2019-11-15:
Your local utility should provide a guide on how to read your meter. If not
try this one: https://southstormont.ca/_uploads/5ent55g4h.pdf   If you're
in the US then the meter is likely calibrated in US Gallons, elsewhere it
is usually in Cubic Meters. The radio transmission itself is 'unitless' -
it simply conveys information that is shown on the face of the meter, based
on the assumption that the utility who is reading it already knows what
physical calibration that particular serial number has.

In terms of the outputs that you're seeing:
  Unkn1: The information in this portion of the data packet can be decoded
into a series of bits, but RTLAMR doesn't know what it's used for yet so
it's labeled as 'unknown' and just shown as a hexadecimal number (for
brevity)
  Unkn3: As with Unkn1, the purpose of this register remains a mystery.
However it, or Unkn1, might indicate stuff like a 'tamper' flag, or a 'low
battery' flag, or some other infrequently used diagnostic feature.
  NoUse: The meter tracks consumption over time internally. If there has
been zero consumption then the meter can report out how roughly long it has
been idle - this is used by the utility to verify that a service that is
supposed to have been shut off hasn't been turned back on unintentionally.
  BackFlow: Similarly the meter can determine the direction of flow of the
water over time. This register shows a value indicating either 'no
backflow', 'intermittent backflow', or 'continuous backflow'. When water
truly flows backwards through the meter it can result in contamination of
the water main so any backflow indication indicates a serious operational
risk. 'Continuous backflow' is usually a special case that indicates the
meter is installed backwards, so the consumption reading will be winding
down, not up, which can create a billing error. This relates to the 'flow'
arrows on the meter display.
  Consumption: As above, this is simply the currently displayed 'dial
units', without the least significant (8th) digit that shows on the meter
display. Assuming gallons, the meter displays down to to 1/10ths of a
gallon so you can see the meter increasing when you're physically looking
at it, and the meter can use the extra precision internally for analysis
(like Backflow and Leak detection), but the utility doesn't need that level
of precision to generate billing so transmitting it over the radio would
just waste energy (remember that the radio is battery powered, with an
estimated life of 20 years, so every coulomb counts),
  LeakNow: As with NoUse, the meter also looks for extended periods of time
when the meter hasn't stopped moving at all and reports it out for
diagnostic purposes. It's also displayed on the face of the meter as the
'dripping tap' icon. It has three states that report out for the last 24
hours; no leak, intermittent leak, and continuous leak. Because the meter
is very sensitive it's possible for a tap left dripping to trigger a leak
indication.
  Leak: This summarizes the LeakNow information over a much larger date
range to indicate if there have been transient leaks, or if there was a
leak that is now resolved, etc.

The meter also buffers the last 96 days of readings, on an hourly basis,
for diagnostic download by the utility. RTLAMR can't access that
information but, if you called to report an anomalous billing, then the
utility could come do a 'manual dump' to see when strange things were
happening - perhaps you were filling a pool, perhaps someone left a
sprinkler on, etc.

There is also more info at
https://www.manualslib.com/manual/1528199/Neptune-E-Coder-R900i.html?page=41

Hope that helps.

closed 2019-12-02 by bemasher