bemasher/rtlamr #131: Work with spread spectrum?

jdtsmith, 2019-11-23:
I'm trying to get a sense if this software-defined radio driver would work with a spread spectrum 900Mhz device, like [this one](https://fccid.io/GIF2012WCE).  I cannot find any information about the messaging protocol for its one-way communication, but perhaps I don't know what I'm looking for.

bemasher, 2019-11-24:
Based on the test report this meter appears to be transmitting FSK packets on 25 different channels. For AMR purposes, channel hopping probably happens on the order of seconds or minutes. An rtl-sdr radio would probably have no trouble listening to it and hopping along.

That said, rtlamr has no current or future plans to support FSK messages.

closed 2019-11-24 by bemasher

jdtsmith, 2019-11-24:
Thanks.  Unfortunately your link is not authorized.  

Taking a closer look at the markings on the transmitter, it seems [this is the relevant](https://fccid.io/GIF2006B/Test-Report/Exhibit-E-Test-Measurement-Report-per-2-1033-b6-739638) FCC report (GIF-2006B).  It's also likely that mine is **not** in fact a spread-spectrum unit, but instead a "narrowband" device that stays at a single frequency (916.4-916.5 MHz).  From the manufacturers of a custom radio board which can read this unit's transmissions, I learned:

> The Badger ORION CE radio protocol is used by Badger ORION Universal and Badger ORION CE water and gas metering endpoints. It is proprietary and encoded, but not encrypted. It comes in two flavors: frequency-hopping and fixed-frequency (“narrowband” in the manufacturer’s terminology). Frequency-hopping models are prominently labeled with the code “FHSS”. 

My transmitter just says "FH".  So, assuming it is a fixed frequency, and that the output is not encrypted but "encoded", does this alter your assessment?

danmandle, 2020-07-30:
@jdtsmith Did you get anywhere with this? I have a Badger Orion and haven't seen its ID show up in RTLAMR.

jdtsmith, 2020-08-05:
No progress to report sadly.