bemasher/rtlamr #133: Using with other SDR radios

noneck35, 2019-12-04:
Is there compatibility with SDRPlay or Hack RF one? Do I need to have RTL-SDR?

Thanks for the help!

parkerlreed, 2019-12-04:
This may help https://www.rtl-sdr.com/rsp_tcp-an-rtl_tcp-compatible-iq-server-for-sdrplay-sdrs/

https://www.reddit.com/r/RTLSDR/comments/1rd3wn/beta_hackrf_tcp_quick_port_of_rtl_tcp_for_hackrf/

noneck35, 2019-12-05:
Thank you very much for this information
Mark

 
 

https://www.reddit.com/r/RTLSDR/comments/1rd3wn/beta_hackrf_tcp_quick_port_of_rtl_tcp_for_hackrf/

bemasher, 2020-01-19:
Rtlamr is designed with rtl-flavored radios in mind. That doesn't mean it couldn't be made to work with other radios, but I don't have time or hardware to support radios other than rtl-sdr's. There are a few mechanisms that are limited such that only rtl-sdr's would work, but they can be easily modified or disabled.

This would be left as an exercise for the user.

closed 2020-01-19 by bemasher