bemasher/rtlamr #141: No rtl_tcp after 'go get github.com/bemasher/rtlamr'

jcwren, 2020-02-11:
First, I know next to nothing about 'go', so I'm probably doing something wrong. But I can generally follow instructions :)

Installed Go with `apt-get install golang-go` on my Debian x86 box, which installed `golang-1.11-go golang-1.11-src golang-go golang-src`

Ran `go get github.com/bemasher/rtlamr` per the instructions. No output, but do have `~/go/bin/rtlamr`. There is also a `~/go/src/github.com/bemasher/rtltcp` directory, but no `rtc_tcp` anywhere under `~/go`. 

Checked `apt-cache search` for `rtltcp`, `rtl_tcp`, `rtl-tcp`, didn't find anything.

So where does `rtc_tcp` come from?

jcwren, 2020-02-11:
D'oh. Never mind, that's part of the `rtl-sdr` package.

closed 2020-02-11 by jcwren

clandestine-avocado, 2021-02-23:
@jcwren I _also_ know next to nothing about Go, and I'm probably doing something wrong too...but I can get `rtlamr` to run. I get a `rtlamr: command not found` error in the terminal.  `rtl_tcp` seems to run fine, and outputs this:

```
$ rtl_tcp
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Tuned to 100000000 Hz.
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
```

I've run `go get github.com/bemasher/rtlamr` per the instructions. No output in the terminal, but I now have `~/go/bin/rtlamr` so I think that worked. Any ideas? Hate to open a new issue when I'm pretty sure I'M the issue.

bemasher, 2021-02-23:
What does the command `which rtlamr` return?

I would guess `~/go/bin` is not in your path.

clandestine-avocado, 2021-02-23:
@bemasher `which rtlamr` runs with no output in the terminal. I've never used Go before, and I've only been using Linux for a few months - so if I'm being dense, bear with me. 

Here is where rtlamr installed:
```
pi@3bp:~/go/bin $ pwd
/home/pi/go/bin
pi@3bp:~/go/bin $ ls
rtlamr  rtlamr-collect

```

**On your advice, running** `/home/pi/go/bin/rtlamr` **appears to get it working.** Is there a way to make it run without specifying the full path every time? If I run just rtlamr from the /home/pi/go/bin folder, it doesn't work - which seems odd to me, but again, I'm a Linux neophyte.

bemasher, 2021-02-25:
You'll probably want to add `$GOPATH/bin` to your `PATH` environment variable. Many OS's use the `PATH` environment variable to search for executables. This is a good resource to learn about setting them: https://superuser.com/questions/284342/what-are-path-and-other-environment-variables-and-how-can-i-set-or-use-them

varnav, 2021-06-07:
I have Raspbian GNU/Linux 10 (buster)

I run this:

```
apt install golang
go get github.com/bemasher/rtlamr
```

apt runs fine, second command gives no output (I guess succeeds too)

Then I run `rtl_tcp` - works fine

Then I run `rtlamr` and get `-bash: rtlamr: command not found`.

`which rtlamr` returns nothing

Command `~/go/bin/rtlamr` works.

mattfox27, 2024-01-21:
> I have Raspbian GNU/Linux 10 (buster)
>
> I run this:
>
> ```
> apt install golang
> go get github.com/bemasher/rtlamr
> ```
>
> apt runs fine, second command gives no output (I guess succeeds too)
>
> Then I run [...]

I get the same -bash: rtlamr: command not found if i run ./rtlamr it works, did you ever figure this out?

bemasher, 2024-01-21:
See https://github.com/bemasher/rtlamr/issues/141#issuecomment-786285171

ubernerdsteve, 2024-09-02:
@mattfox27 I know this is old, but I just stumbled across this. If you didn't figure out what he means with add "$GOPATH/bin to your PATH environment variable", essentially you need to set terminal configurations to know where to look, that is the .bashrc file.

So this should fix your issue and save you the googling legwork of trial and error of something you don't quite follow; here is the cut/paste solution for you (and anyone else stuck with this) to try and figure out what this all means (though you should still read the document @bemasher linked to actually better understand what is actually happening here).

`sudo nano  ~/.bashrc`

Go to the end of the document and insert:
`#GOPATH
export GOPATH=$HOME/go
export PATH=$PATH:$GOROOT/bin:$GOPATH/bin
`
Then reboot.
'sudo reboot'

Should work after that.  Depending on your linux distro some syntax might be slightly different.