bemasher/rtlamr #146: SCM last 3 digits = Revolutions?

ShoGinn, 2020-03-22:
Okay, so I've been just going crazy here trying to figure out why the numbers on my setup are off..

Im using rtlamr-collect to feed influx.. but thats not the issue.

Here is a line...
```
16:55:32.469391 main.go:458: rtlamr,endpoint_id=513XXXX,endpoint_type=5,msg_type=cumulative,protocol=SCM consumption=8762033i 1584910532469179246
```
If you notice the consumption is 8762033
When looking at the meter:
[image]

It is at 3.8? kWh?

I went outside with the dump of the log running, and noticed that the last 3 digits increment every revolution.

I've tried to figure out the way to make this work but can't.

It gets to 8761999 then rolls over to 8762000

I've tried to multiple revolutions by 7.2 (listed on the meter) but that doesn't net the correct result.

33 * 72 = 237.6 watts (0.2376 kWh)

What am I missing?

ShoGinn, 2020-03-22:
For further data points:

362 | almost 7
-- | --
175 | just over 5
143 | almost 5
82 | just over 4
33 | almost 4

ShoGinn, 2020-03-22:
[image]

bemasher, 2020-03-22:
Can you provide historical data from your meter in json format?

ShoGinn, 2020-03-22:
https://gist.github.com/ShoGinn/caf8d0906345912fb2589d748c85fbf4

bemasher, 2020-03-22:
It's not uncommon for meters to report more digits of precision through ERT than they display, even on analog meters. It appears that those last three digits are just extra digits beyond the decimal.

ShoGinn, 2020-03-22:
Thats weird.. like I said. I diagnosed it to mean revolutions (1-999)

I watched the log, when it would increment, it was at the 100 part of the wheel

ShoGinn, 2020-03-22:
Well im at 8762682 right now and the meter shows 87630 very odd ;)

ShoGinn, 2020-03-23:
So the problem is far more of an issue. Because the 10kWh dial gets incremented on the analog meter at the 650 range. That means that it is reporting significantly less than actually on the meter.
Could this be a parsing error with the signal?

ShoGinn, 2020-03-28:
Well I figured out the correlation.

![(10*Consumption)+3500](https://render.githubusercontent.com/render/math?math=(10*Consumption)%2B3500)

Not sure why they use this..

I was bashing my head against the wall wondering why the 7.2 watt-hour per repo didn't match the meter.

But this is much cleaner since you can take the consumption value and just turn it into KWH.

ex.

![Consumption = 8783091](https://render.githubusercontent.com/render/math?math=Consumption%20%3D%208783091)

![(10*8783091)+3500 = 87834410 kWH](https://render.githubusercontent.com/render/math?math=(10*8783091)%2B3500%20%3D%2087834410%20kWH)

And the meter shows:

[image]

bemasher, 2020-03-30:
Nice! Glad you were able to sort that out.

closed 2020-03-30 by bemasher

ShoGinn, 2020-03-30:
Anything that is an scm feature you can use? Or is this unique?  

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

bemasher, 2020-03-30:
This appears to be unique. This isn't the first time I've seen meters transmitting oddly encoded consumption values. There are some R900 water meters in the wild that transmit binary-coded digits rather than the usual consumption amount.

SCM is primitive enough that you can't really tell much about the meter you heard a message from with only the message. SCM+ and IDM both transmit a bit more information you can use to determine what to do with your meter's messages, but otherwise, you're kind of on your own.