bemasher/rtlamr #147: No data packets being received (I think?)

dreday99, 2020-04-22:
Hi, I am going to apologize beforehand, as I am pretty inexperienced with source coding and coding in general.  I believe I am running all the steps correctly, however, I don't seem to be receiving any data on my RTLAMR terminal window.  I am running OS X 10.13.6.  My water meter is a SENSUS iPERL.  Here are all the steps as well as the command results:

1. First I run rtl_tcp in terminal window 1:
[image]

2. Next, I open a second terminal window and run rtlamr:
[image]

3. This is what the rtl_tcp window looks like once I run RTLAMR:
[image]

I have had rtlamr run for about 24h and nothing ever popped up on my rtlamr terminal in terms of data packets.  Is there anything I am doing noticeably wrong?  Thank you

dreday99, 2020-04-22:
I want to point out that the blank spaces in the code above was just me removing my last name from the line.  Thanks.

bemasher, 2020-04-23:
Sensus meters are generally not supported. You're not seeing messages because your meter likely uses a protocol other than ERT.

closed 2020-04-23 by bemasher

dreday99, 2020-04-24:
Is there any way to support it or is this a complete dead end for me?

Thanks again.

bemasher, 2020-04-28:
Sensus is quite tight-lipped about their protocol and documentation for it, without physical access to a provisioned meter, it would be more difficult to reverse-engineer than I have time or resources for. Sorry.