bemasher/rtlamr #151: GE I-210+ line of smart meters

claud9999, 2020-05-27:
Anyone have success reading from the GE I-210+ line of meters?

https://www.aclara.com/products-and-services/smart-meters/ansi-residential/i-210-and-i-210c/

I have my RTLAMR software running but am not seeing any messages, yet.

natwilson, 2020-06-21:
I have the same meter and also been unable to decode any readings from over the air.  GE makes the meter portion and they offer a few network card options built by other companies.  My utility, Dominion, uses the Silver Spring Networks model NIC 511.  I believe it's encrypted based on publicly available documents:
https://www.gedigitalenergy.com/products/brochures/i210_with_SSN_NIC.pdf

These meters offer the utilities dozens of configuration options and the firmware is remotely upgradable so your experience may be different.

I've still been able to read the meter electronically by building an IR pulse reader, but it's not nearly as elegant.

timmhayes, 2020-06-21:
@natwilson My utility is also Dominion. A few years ago they replaced all the old meters in my neighborhood with ones using the Silver Spring Networks model NIC 311. I'm new to this, but confused how I can pick up 15-17 local meters...but not mine. I should also say my meter is displaying an error of 000200, but when I called Dominion about it they said the meter is still reading correctly. I was hoping they would replace it with one I could read.

MACscr, 2020-06-25:
Hmm, i have the I-210+ from Florida Power & Lighting (FPL) from Silver Springs. It has a Nic 314. I haven't even bought any hardware yet to try to read it as I am trying to verify it can even be read and what information I might be able to get out of it over what I am already from a python library that provides basic info such as:
``` print(api.yesterday_kwh)
    print(api.yesterday_dollars)
    print(api.mtd_kwh)
    print(api.mtd_dollars)
```

Hoping to get more detailed usage from reading the meter directly.

MACscr, 2020-06-25:
Looking at this #108 discussed earlier, we might be out of luck

bemasher, 2020-06-26:
Silver Springs-based meter transceivers are likely never going to be supported.

closed 2020-06-26 by bemasher