bemasher/rtlamr #153: Add info on identifying your meter

skullydazed, 2020-06-11:
I don't know how universally true this is, but I found that my particular meter was easy to identify once I noticed that the ID is printed just above the KWH display. I think this would be useful information to include in the Ethics section of the readme.

[image]

bemasher, 2020-06-15:
There are a bunch of meter photos available in fcc test reports, I'll compile some examples.

mmiller7, 2021-07-18:
I'll add this may vary a lot by meter type, in my case (Dominion Energy analog meter) the ID on the front of the dial is wrong, the correct ID to look for is on a label affixed to the RF module under/behind the dial visible thru the bottom of the glass.

`{Time:2021-07-18T12:26:33.198 SCM:{ID:48722275 Type: 5 Tamper:{Phy:03 Enc:03} Consumption: 2127374 CRC:0xB011}}`

[image]

commit 3965b5c5 (2021-07-18): Add information about identifying your meter. Fixes #153.
closed 2021-07-18 by bemasher