bemasher/rtlamr #155: identifying IDM packet for Net meters

evilpete, 2020-06-26:
How do you detect is a IDM packet is  for Net Meters if the both start with the same id data 

Preamble     0x5555 
SyncWord  0x16A3 
ProtocolID  0x1C 
PacketLength  0x5C 
HammingCode   0xC6

bemasher, 2020-06-26:
Net meters typically have an endpoint type of 8, while non-net meters have type 7.

closed 2020-06-26 by bemasher

evilpete, 2020-06-26:
That is not what I am observing in the samples i have

{Time":"2020-06-25T08:22:08.569276915-04:00","Offset":1605632,"Length":229376,"Type":"**NetIDM**","Message":{"Preamble":1431639715,"ProtocolID":28,"PacketLength":92,"HammingCode":198,"ApplicationVersion":4,**"ERTType":7,**"ERTSerialNumber":1550406067,"ConsumptionIntervalCount":30,"ProgrammingState":184,"LastGeneration":125,"LastConsumption":0,"LastConsumptionNet":2223120656,"DifferentialConsumptionIntervals":[7695,545,2086,1475,6240,2180,4240,4616,240,7191,609,7224,1603,96,2052,12464,6152,8480,9226,352,12312,833,10292,1795,4248,4613,8416],"TransmitTimeOffset":2145,"SerialNumberCRC":61178,"PacketCRC":37271}}

{"Time":"2020-06-25T08:22:52.404629556-04:00","Offset":1835008,"Length":229376,"Type":"**IDM**","Message":{"Preamble":1431639715,"PacketTypeID":28,"PacketLength":92,"HammingCode":198,"ApplicationVersion":4,**"ERTType":7**,"ERTSerialNumber":11278109,"ConsumptionIntervalCount":246,"ModuleProgrammingState":188,"TamperCounters":"QgUWry0H","AsynchronousCounters":0,"PowerOutageFlags":"QUgmCEEF","LastConsumptionCount":339972,"DifferentialConsumptionIntervals":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,1,0,0,0,0,0,0,0,0,0,1,0,0],"TransmitTimeOffset":476,"SerialNumberCRC":60090,"PacketCRC":31799}}

{"Time":"2020-06-25T08:23:10.23581967-04:00","Offset":2064384,"Length":229376,"Type":"**NetIDM**","Message":{"Preamble":1431639715,"ProtocolID":28,"PacketLength":92,"HammingCode":198,"ApplicationVersion":4,**"ERTType":7,**"ERTSerialNumber":1550406067,"ConsumptionIntervalCount":31,"ProgrammingState":184,"LastGeneration":125,"LastConsumption":0,"LastConsumptionNet":54673680,"DifferentialConsumptionIntervals":[8719,481,1060,1218,14444,3076,4232,4617,256,7694,737,3118,1795,2144,3076,152,5644,272,9234,320,11288,769,10292,1667,8296,4873,176],"TransmitTimeOffset":732,"SerialNumberCRC":61178,"PacketCRC":40443}}

{"Time":"2020-06-25T08:26:09.784225228-04:00","Offset":2752512,"Length":229376,"Type":"**IDM**","Message":{"Preamble":1431639715,"PacketTypeID":28,"PacketLength":92,"HammingCode":198,"ApplicationVersion":4,**"ERTType":7,**"ERTSerialNumber":13897632,"ConsumptionIntervalCount":172,"ModuleProgrammingState":188,"TamperCounters":"ggOADwgg","AsynchronousCounters":0,"PowerOutageFlags":"oAQAAAAA","LastConsumptionCount":1436228,"DifferentialConsumptionIntervals":[3,2,3,3,3,2,3,3,2,3,5,7,7,5,6,6,2,4,5,4,2,3,4,4,4,4,4,10,10,10,10,10,10,9,9,9,9,10,9,9,9,9,9,10,9,9,9],"TransmitTimeOffset":1390,"SerialNumberCRC":26559,"PacketCRC":31002}}
: