bemasher/rtlamr #156: Compatibility question - Landis+Gyr Focus kWh

mrd0n, 2020-07-26:
Hi - curious if the Landis+Gyr Focus KwH electrical unit is supported or if you think it could be?

The FCC ID is :
https://fccid.io/ROV-CLTR900M

The manual for the unit is:
https://www.solar-electric.com/lib/wind-sun/Focus-technical-manual.pdf

Thank you for this awesome contribution - my water meter and gas meter are being captured perfectly!

mrd0n, 2020-07-29:
Adding some more info that is hopefully useful.

From the FCC document, the device frequency range is 913.98-917.58 MHz.

I've tried a capture using
`rtlcap -centerfreq=915M -tunergainmode=false -gainbyindex=29 -squelch=0.658 -bytes=25M -o capture-915M.txt`

I think the device ID is 40615079.

Added some photos.
[capture-915M.txt](https://github.com/bemasher/rtlamr/files/4992178/capture-915M.txt)
[image]
[image]

bemasher, 2020-08-03:
Your 100G should be compatible based on the label, but your Focus is likely not compatible. Based on the samples you provided, it does not contain any messages from your gas meter.

Below is a sample test to decode the samples you provided.

```go
package main

import (
	"io"
	"os"
	"testing"

	"github.com/bemasher/rtlamr/idm"
	"github.com/bemasher/rtlamr/protocol"
	"github.com/bemasher/rtlamr/scm"
)

func TestRead(t *testing.T) {
	sampleFile, err := os.Open("capture-915M.cu8")
	if err != nil {
		t.Fatal(err)
	}
	defer sampleFile.Close()

	d := protocol.NewDecoder()
	d.RegisterProtocol(scm.NewParser(72))
	d.RegisterProtocol(idm.NewParser(72))
	d.Allocate()

	block := make([]byte, d.Cfg.BlockSize2)

	for {
		// Read new sample block.
		_, err := sampleFile.Read(block)
		if err == io.EOF {
			break
		}
		if err != nil {
			t.Fatal("Error reading samples: ", err)
		}

		offset, _ := sampleFile.Seek(0, os.SEEK_CUR)

		for msg := range d.Decode(block) {
			t.Logf("%9d %T %+v\n", uint64(offset)>>1, msg, msg)
		}
	}
}
```

This is the output.

```
=== RUN   TestRead
    main_test.go:40:    253952 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    270336 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    303104 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    319488 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    352256 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    376832 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    393216 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    417792 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    434176 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    450560 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    475136 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    491520 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    516096 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    548864 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    565248 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:    598016 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
[... 36 lines cut ...]
    main_test.go:40:   1343488 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1359872 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1376256 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1392640 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1409024 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1441792 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1466368 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1499136 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1515520 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1531904 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1548288 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1564672 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1581056 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1597440 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1613824 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1638400 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1662976 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1687552 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1720320 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1736704 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
    main_test.go:40:   1761280 scm.SCM {ID:30037481 Type:13 Tamper:{Phy:00 Enc:03} Consumption:  149960 CRC:0x89E9}
--- PASS: TestRead (0.26s)
PASS
ok  	github.com/bemasher/rtlamr/data/focus	(cached)
```

closed 2020-08-03 by bemasher

mrd0n, 2020-08-03:
Thank you for your time!   Bummer on the Focus not likely being compatible.    

The 100G gas meter is being decoded - yay!.

seancolyer, 2021-02-27:
@bemasher do you know what the issue is here? Looking at a couple of the other tickets it looks like there's a variety of potential issues that seem to group around two classes:
* Signal being different type (FSK/variable packet length)
* Signal being encrypted

Is that right? If it's the first issue, I was thinking about trying to dive in and see if I could possibly put something together built on your work, would that make sense to attempt?

bemasher, 2021-02-27:
I have no direct experience with non-ERT meters, you are welcome to fork rtlamr and build support yourself. I don't have time or resources to support these meters in rtlamr, so it would have to remain a separate project.

bemasher, 2021-05-27:
For the curious among you, this page has quite a lot of good information regarding Landis+Gyr meters and decoding the protocol at least some of them use.

https://wiki.recessim.com/view/Gr-smart_meters_Setup_Guide