bemasher/rtlamr #157: Focus AXR SD NetMeter

gkankanh, 2020-08-02:
I have a Focus AXR SD NET meter on Silver Spring Networks with FCCID OWS-NIC514 . We have solar feeding the grid back.
When I run rtlamr I do see some decoded packets for SCM only.
{Time:2020-08-02T19:41:01.653 SCM:{ID: 5850975 Type: 4 Tamper:{Phy:00 Enc:01} Consumption:    6056 CRC:0x3065}}
{Time:2020-08-02T19:41:09.321 SCM:{ID: 5850975 Type: 4 Tamper:{Phy:00 Enc:01} Consumption:    6056 CRC:0x3065}}
{Time:2020-08-02T19:41:13.377 SCM:{ID: 5850975 Type: 4 Tamper:{Phy:00 Enc:01} Consumption:    6056 CRC:0x3065}}
{Time:2020-08-02T19:41:22.488 SCM:{ID: 5850975 Type: 4 Tamper:{Phy:00 Enc:01} Consumption:    6056 CRC:0x3065}}
{Time:2020-08-02T19:41:25.432 SCM:{ID: 5850975 Type: 4 Tamper:{Phy:00 Enc:01} Consumption:    6056 CRC:0x3065}}

I do not think these readings correlate to my Electricty meter since the consumption is increasing linearly always.
I read that this particular meter uses FHSS/FSK and may even encrypt the payloads. My Questions are as follows
1. Is my meter forwarding / proxying for my neighbor in the clear without encryption?
2. Any tips on how I can find my meter id?
Thanks for rtlamr. Fantastic work. hats off!.

bemasher, 2020-08-03:
Rtlamr does not support any devices with Silver Spring Networks radios. Sorry.

closed 2020-08-03 by bemasher

fgonza2, 2022-05-18:
hi. has this changed? i have one of those meters as well, and was wondering if there is a way to read it