bemasher/rtlamr #160: OpenWay by Itron meter?

maurera, 2020-12-15:
Hi @bemasher - thanks very much for putting this project together. I'm interested in trying it out with my electric and/or water meter. Could you suggest whether this is likely compatible with my meters? If so, I'll go ahead and buy a rtl-sdr dongle.

**My electric meter**
"OpenWay by Itron", model AMI7. I've found an [fcc document](https://fccid.io/SK9AMI7/User-Manual/Manual-1459875) that states
```
The Itron AMI7 is an electricity metering module which includes the register and display features for the meter as well as a Zigbee transmitter that operates in the 2405 MHz to 2475 MHz band and a LAN transmitter that operates in the 902 MHz to 928 MHz band.
```
[(picture)](https://imgur.com/a/4rcwgWV)

**My water meter**
"Neptune Auto G65N 5/8 T-10". 
I was under the impression all water meters in my city were smart meters, but I'm not sure this is actually a smart meter? I wasn't able to figure it out from google searches
[(picture)](https://imgur.com/a/s7W2oeW)

TrendMend, 2020-12-16:
I don't think yours is a smart meter, if you look at the description from
the site below it states that it doesn't have remote reading capabilities
(it's not smart).

https://www.flows.com/neptune-t-10-3-4-potable-water-meter/

bemasher, 2020-12-16:
rtlamr does not support Itron's OpenWay.

closed 2020-12-16 by bemasher

DATAstrm, 2022-02-17:
Hello, is there any plan to attempt to develop compatibility with Openway?  I think I saw somewhere you mentioning that you were not doing that due to lack of accessibility to testing products.  I'd be happy to help with my meter, if that would start things up.  Thanks!

bemasher, 2022-02-17:
There are no plans to implement OpenWay because it is likely encrypted, and I don't have time or equipment to implement an entirely different protocol.

DATAstrm, 2022-02-17:
Any way those in the community can help with equipment or research?  Loving the way rtlamr is working for the gas meter and it would be great to have wider support!

Best regards,
Guang

bemasher, 2022-02-17:
Feel free to fork this project for your own implementation, I don't have time to work on implementing OpenWay.