bemasher/rtlamr #161: Compatibility

boyacimustafa, 2020-12-20:
[image]

First of all thank you for creating such an awesome project.
I can a reading from my electric meter with default settings but I couldn't get any data from my water meter.
Above is the picture of my water meter communication module. Is it a 100WD or 100W+? I see 100WD in the compatibility list but If it is 100W+, am I out of luck to get any reading from it? Thank you.

bemasher, 2020-12-23:
Based on the barcode, your meter's type should be 11 and ID should be 79349223. The FCC test report indicates the meter supports both AM and FM modes, which will determine whether it is supported by rtlamr. I would guess that it is supported though.

bemasher, 2021-02-18:
There is also the possibility it is transmitting Network Interval Messages, which I believe are FSK modulated: [Itron Water AMI 100W Communication Module Network Interval Message](https://www.itron.com/-/media/feature/products/documents/white-paper/water-ami-100w-network-interval-message.pdf)

closed 2021-02-24 by bemasher