bemasher/rtlamr #170: Itron ERT Model 40G gas meter

Adminiuga, 2021-02-05:
Hi,
Looks like I have an Itron ERT Model 40G gas meter. I don't see any FCC ID on the outside labeling, but I was able to catch it reading which I suspect are happening on demand, as while running the `rtl_collect` continuously, i'm getting those readings only in the span of a few minutes, roughly every other week.
Can anyone confirm those could be configured to broadcast readings only on demand and if there are any pointers on the signal/trigger?

bemasher, 2021-02-18:
It is pretty likely that your silent meter is operating in "wake-up" mode, which means it will only transmit when asked to by the utility.

Adminiuga, 2021-02-18:
Yep, looks so. Just wonder if anyone tried ti reverse the wake-up signal? gegt 
Since I only get the readings once or twice a month, makes a bit problematic to catch the actual signal.

closed 2021-02-18 by Adminiuga

bemasher, 2021-02-18:
See this issue for reasons you probably wouldn't want to wake the meter yourself.

https://github.com/bemasher/rtlamr/issues/103#issuecomment-425316891

Adminiuga, 2021-02-18:
That valid concern. Was not expecting the wake up signal be outside of ISM band.

bemasher, 2023-08-13:
Also see #105