bemasher/rtlamr #175: Research information

hllhll, 2021-02-13:
Hello,
Here we have a similar product, I guess. it uses 433 ISM band, with some form of FHSS and a 4fsk or c4fm or CPSM 4 whatever you'll call it. Also i've found links between the company issued the tech here, and iTron.
[HackAday specifies](https://hackaday.com/2014/02/25/using-sdr-to-read-your-smart-meter/) that there's some paper explaining how you researched everything, but it's 404.

1. Is there any chance it would just work?
2. where can I find the paper/research information/how did you solved the tens of different issues that it takes.

Personally, I've spend weeks trying just to decode a short 4fsk burst, only to realise it's combined within a wider burst pattern and I've succumbed to my fatigue trying solve this project. 
The meter can be triggered to send data with a magnet somewhere along the meter (water meter).

bemasher, 2021-02-18:
There's no chance that could "just work" with rtlamr. ERT is a manchester coded OOK signal.

closed 2021-02-18 by bemasher

hllhll, 2021-02-18:
And about the title of the question, what about (2)? Do you have some documentation/thought process how you got to this implementation and/or, how do you  track fhss?

bemasher, 2021-02-24:
Rtlamr doesn't track with the meter when it hops channels. Except for IDM packets, the data transmitted doesn't update with every hop anyway, so it's good enough to listen to about 12 contiguous channels. IDM packets only update a time field every transmission.

Luckily, ERT's modulation and coding make it very straight forward to demodulate and decode messages, even very weak signals when listening to a relatively wide portion of the band they are transmitted in.