bemasher/rtlamr #187: rtlamr High CPU usage

sprior, 2021-04-16:
A friend and I are noticing that rtlamr is taking up 25% of a CPU making it by far the highest load on the machine, which seems like a lot given what it does, memory usage seems fine.  I don't know it it's related, but we're both using the unique mode.

I'm not familiar with the Go language, but by any chance is there polling going on instead of blocking on input from rtl_tcp?

bemasher, 2021-04-17:
What hardware are you running rtlamr on and which message types are you decoding?

Communication with `rtl_tcp` is likely not a botte-neck in rtlamr's performance. If you're curious, the goroutine for obtaining samples from `rtl_tcp` is found here: https://github.com/bemasher/rtlamr/blob/master/main.go#L155-L192

Signal processing in software requires a non-trivial amount of floating point operations. Listening at the highest reliable sample rate on rtl-sdr dongles requires processing 2.4M quadrature samples every second. Determining if a message is present in received samples requires demodulation, filtering, bit-decision, and a preamble search.

 * Demodulation converts 2x 8-bit unsigned integers to 1x 64-bit float.
 * A matched filter is applied to the incoming 64-bit samples. For the R900 decoder, 3x matched filters are applied.
 * Bit-decision reduces the incoming 64-bit floats to 1-bit samples packed into 8-bit integers.
 * Preamble search returns only indices into the 8-bit integers where a valid preamble is found.

At the expense of sensitivity, some of these operations could be reduced by down-sampling, but I have found that rtlamr is efficient enough that this is not necessary.

sprior, 2021-04-17:
I'm running rtl_tcp (and nothing else) on a Raspberry Pi 2 and running rtlamr  on one virtCPU of an Intel Xeon in a Docker container.  But I'm new at SDR so I didn't realize how much of the load was being done on the CPU so I need to adjust my expectations.  I thought that rtl_tcp was doing the radio work.

bemasher, 2021-04-17:
rtl_tcp is a spectrum server. It provides a control channel for the radio and serves up samples.

sprior, 2021-04-17:
That sounds like something you don't necessarily want clogging up your main network.

Nils154, 2021-04-17:
You can run both rtl_tcp and trl_amr on a pi2, but don’t expect it to do anything else.

sprior, 2021-04-17:
If I can reasonably run rtl_tcp, rtlamr | rtlamr_collect and have it post to Influx on a Pi2 which is running RaspberryOS Lit e(no desktop) then I'd be totally happy.

bemasher, 2021-04-17:
I set up a similar configuration for my dad, RPi4 with the raspberry pi release of debian, rtlamr, rtlamr-collect, influxdb, and grafana all on the same pi. Though I would recommend connecting a more durable storage device than the microSD. His setup has a 120GB SSD connected by USB and has influxdb's data files stored there.

sprior, 2021-04-17:
Oh the Influx (and Grafana) I was referring to was hosted on another machine.  Is it reasonable to run ONLY rtl_tcp, rtlamr | rtlamr_collect on a Pi2 and nothing else?

Nils154, 2021-04-17:
That worked for me, but I had to specifically request SCM and R900, it would not decode the R900 messages listening to all types.

bemasher, 2021-04-18:
Be careful with the R900 decoder on pi-based systems, it requires a lot more processing relative to the ERT-flavored messages.

bemasher, 2021-04-18:
> Oh the Influx (and Grafana) I was referring to was hosted on another machine. Is it reasonable to run ONLY rtl_tcp, rtlamr | rtlamr_collect on a Pi2 and nothing else? No X server and no data storage on the pi.

That is the recommended use-case. My setup is an RPi 3 with rtlamr and rtl_tcp feeding an influxdb instance on another system. The RPi handles a few other telemetry related services, but otherwise that's it.

sprior, 2021-04-18:
Ok great, I have enough info to keep working on this, thanks.

closed 2021-04-18 by sprior