bemasher/rtlamr #189: Once per day data collection

TrevorJMarshall, 2021-04-28:
Hi Douglas-

Thank you for your rtlamr contribution.

I recently purchased a Nooelec NeSDR and have been successfully reading my gas and electric meters using your rtlamr software with the Debian Linux operating system.

For a few days I used the suggested two terminal windows, but have moved to shell scripts run from a single terminal as follows:
 
(gas meter script gmeter.sh)

rtl_tcp &
sleep 6
go/bin/rtlamr -filterid=xxxxx -single=true -msgtype=scm+ | tee -a gmtr
sleep 6
pkill -2 rtl_tcp
wait
exit

(script for electric meter similar)

With shell scripts the rtl_tcp process needs to be killed with a SIGINT (2) signal (equivalent to ctrt-C in Terminal) otherwise the dongle remains on and becomes quite warm. I only wish to take readings once a day.

My next step is automatic daily logging, so my question is:

Would your rtlamr-collect logging software be adaptable to this once a day dongle power-up scheme or should I develop my own method using cron, shell scripts and so forth? 

I imagine other people have similar needs.

Best regards, Trevor Marshall

bemasher, 2021-04-28:
`rtlamr-collect` does take input from rtlamr and maintains some state to avoid storing repeated messages, but it is intended to be run all the time, same with rtlamr.

TrevorJMarshall, 2021-04-29:
OK, thank you. You answered my question. I will attempt to use rtlamr-collect as an interface between rtlamr and influxdb. Do I need to report back on success, or should I "Close with comment" (Git messaging neophyte here)

bemasher, 2021-04-30:
You can close it if it resolved the issue. You can always come back later and comment your results on this issue.

closed 2021-04-30 by TrevorJMarshall