bemasher/rtlamr #194: Landis & Gyr Focus AXR-SD Power Meter

MauriceB13, 2021-06-30:
I understand that this has come up in the past and there seems to be quite a demand for RTL-SDR support for the Landis & Gyr Focus AXR-SD meter.

I have a Landis+Gyr FOCUS AXR-SD FORM 2S CL200 240V 3W 60Hz TA=30 Kh 7.2 Trilliant Networks RES-30000-FocusAX Securemesh 1w FCC: TMB-G30Focus IC: 6028A-G30Focus.

I have attempted using RTL_TCP and RTLAMR and RTLCAP to see if my meter can be picked up without success.

I would really like to figure this out and be able to read the data from this meter.

I am willing to do whatever is needed to facilitate this with your guidance @bemasher. Could we collaborate on this?

Thanks,
Ray

### Discussed in https://github.com/bemasher/rtlamr/discussions/191

<div type='discussions-op-text'>

<sup>Originally posted by **thomastech** April 20, 2017</sup>
Thank you for sharing your project! I'd like to integrate it with my OpenHabian system to monitor power usage. But for initial testing I installed it on a Win10 Intel desktop.  

With SDR# I can see/hear brief data chirps every 17 seconds at 914.889MHz and 915.040MHz (simultaneously).   But no data is dumped while in rtlamr. 

I've experimented with the msgtype and tried all 5 formats. I also tried centerfreq 915040. No luck so far.

My power meter is a Landis & Gyr Focus AXR-SD, FCC ID OWS-NIC507. After some searching I found the github discussion in closed ticket #55 that explained this meter is not supported if it is using an encrypted communication method.  A confirmation would require checking the FCC ID info.

I checked the FCC database and could not find any reference to the communication protocol. Do you mind looking to see if there is hope of adding it as a supported meter? Here's a direct link:
https://fccid.io/OWS-NIC507

- Thomas</div>

TheChrisTech, 2021-07-01:
Interesting. I was just looking at this tonight, and have a similar module (different FCC and IC).

Yours (which includes a manual): https://fccid.io/TMB-G30FOCUS
Mine: https://fccid.io/SDBLGZ1000

One difference, is I see that mine comes with Zigbee - I'll need to see if I can get my hub to connect with it.

bemasher, 2021-07-01:
Like the previous issues regarding this and other similarly flavored meters, support will not be added to rtlamr.

See https://wiki.recessim.com/view/Main_Page for information about possibly listening to these meters.

closed 2021-07-01 by bemasher