bemasher/rtlamr #223: r900 consumption value differs?

tyjlin, 2021-12-02:
Just getting the rtlamr running and trying out the r900 decoder. I was able to find and decode based on my water meter ID. However, I noticed that there seems to be offset between the reported consumption level from the r900 vs the actual reading from the water meter.

Meter read would be at: 3753146.45
Meter read using r900 is at: 3977032

I read somewhere that the Naptune 900i can potentially be customized by the utility company, but seen as the two values are relatively close, I wonder if there was a minor issue with the decoder? Sorry I don't know go, so I can't really provide too much more info, but if there are any test you wish to run, I can try to help out.

eea123, 2022-02-05:
Interesting.  I would think that the offset meter reading might have been added to account for a replacement meter at some point in time.  Usually, the utility would just do a final read on the old meter and start with a zero read on the new one unless they swapped a used one in.  Still, that should be handled by the billing software back their office rather than by the radio in the field.

bemasher, 2022-05-10:
Closing due to age.

closed 2022-05-10 by bemasher