bemasher/rtlamr #227: No message received

3dinhjeffrey, 2021-12-22:
Running on Raspberry pi 3b+

using 2 terminals, i ran 

# Terminal A
`$ rtl_tcp`

# Terminal B
`$ rtlamr`

i also tried with `rtlamr -msgtype=all -centerfreq=915M -samplerate=1382400`

on terminal A, I would get ll+, now 1 (all the way up to 502, then restarts)
on terminal B, nothing shows up.

Any advice? Thanks
[image]
[image]

bemasher, 2021-12-23:
A couple of things come to mind:

1. You probably don't have any rtlamr compatible meters.
2. Using `-msgtype=all` is likely causing rtl_tcp to build up linked lists of unconsumed samples because it includes the r900 decoder which is significantly less efficient than the other decoders and rtlamr isn't consuming samples fast enough.
3. You might try without the r900 decoder, and you may pick up some messages that were otherwise being missed.

3dinhjeffrey, 2021-12-23:
@bemasher Thank you for your response, really appreciate it

I ended up doing `rtlamr -msgtype=scm,scm+,idm -centerfreq=915M -samplerate=1382400`

I didn't get `ll+` messages anymore. but also didn't receive any meter messages. 

Couldn't find my meter ERT Type, but both my electricity and gas are ltron.

Electricity:
[image]

Gas:
[image]

Drizzt321, 2022-02-14:
I'm having a similar issue, don't see any messages coming in. Same exact meter, down to the FCC ID. Based on my reading from [this comment](https://github.com/bemasher/rtlamr/issues/83#issuecomment-352597248), it's sounds like my meter (mine is SoCal Edison) uses OpenWay and is probably encrypted? Or do I need to iterate through various frequencies to try and find what it's transmitting on?

```
$ go/bin/rtlamr -msgtype=scm,scm+,idm
14:35:46.044707 decode.go:45: CenterFreq: 912600155
14:35:46.044983 decode.go:46: SampleRate: 2359296
14:35:46.044991 decode.go:47: DataRate: 32768
14:35:46.044997 decode.go:48: ChipLength: 72
14:35:46.045002 decode.go:49: PreambleSymbols: 32
14:35:46.045007 decode.go:50: PreambleLength: 4608
14:35:46.045012 decode.go:51: PacketSymbols: 736
14:35:46.045017 decode.go:52: PacketLength: 105984
14:35:46.045042 decode.go:59: Protocols: scm,scm+,idm
14:35:46.045052 decode.go:60: Preambles: 111110010101001100000,0001011010100011,01010101010101010001011010100011
14:35:46.045070 main.go:126: GainCount: 29
```

bemasher, 2022-02-16:
Feel free to try different frequencies, but most Itron meters I've seen use frequency hopping so you're pretty likely to hear something on the default frequency. If it's OpenWay, you're out of luck, this project doesn't support that.

closed 2022-02-16 by bemasher

mwaterbu, 2022-06-06:
> If it's OpenWay, you're out of luck, this project doesn't support that.

Thank you for this information (now after searching I see that you have mentioned this on many issues over several years). After checking the list of [compatible meters](https://github.com/bemasher/rtlamr/wiki/Compatible-Meters), I was initially under the impression that mine would be supported, as I was blissfully unaware that OpenWay existed.

I had since tried a couple different USB RTL-SDR units, and various antennas, confused as to why I wasn't receiving any consumption messages, however, it appears this is because I probably don't have any rtlamr compatible meters.

@bemasher if you are aware of some meters which could be configured with OpenWay, such as the C2SOD, do you think it could make sense to mark them with an asterisk on the Compatible Meters page in the Wiki (with an explanation stating that if the meter has the OpenWay logo, it probably isn't going to work)? It may save some issues from being opened, as well as user's time.