bemasher/rtlamr #244: Arad / Master Meter water utility meter

avicarmeli, 2022-03-20:
I am trying to add a decoder for Arad/Master Meter water utility meter to rtl433 with some success. I was able to read meter's serial number and water counter as I posted [here](https://reverseengineering.stackexchange.com/questions/30133/reverse-engineering-a-weird-32-bit-possibly-not-crc-checksum). 
I wonder, maybe with your experience with other meters you can help decoding the rest of the massage?

bemasher, 2022-04-27:
From your captures, can you tell me what the modulation is?

avicarmeli, 2022-04-27:
It is Manchester NRZ IEEE.

bemasher, 2022-05-10:
Hmm, can you provide some sample captures?

avicarmeli, 2022-05-10:
Do you mean other than what I posted [here](https://reverseengineering.stackexchange.com/questions/30133/reverse-engineering-a-weird-32-bit-possibly-not-crc-checksum).?

bemasher, 2022-05-20:
I mean IQ data, if possible.

avicarmeli, 2022-05-20:
You can find [here ](https://drive.google.com/drive/folders/1o2rRn8eZM9zINk7zcZ138HbxPmDh7y2G?usp=sharing)some data.

nickrobinson, 2022-08-12:
@avicarmeli Were you able to make any progress here? I'm working through trying to get data from a Master Meter and am trying to replicate some of your work here.

avicarmeli, 2023-01-13:
@nickrobinson you can take a look at what I have by now  [here](https://github.com/avicarmeli/rtl_433).
It is still a work in progress.

Mindful, 2023-07-26:
@avicarmeli Are you still working on this? 
I've suddenly become very interested in being able to read my (Master Meter) water meter.

avicarmeli, 2023-07-26:
Hi Josh

I ended up just compiling very limited version of the driver since I can only read the water consumption and could not decode the rest of the massage.

It works fine for me to read the water counter.

If need any specific help don't hesitate to write me.

Avi

avicarmeli, 2023-07-26:
One more thing is that I might made the preamble longer in order to read only my meter , so in that [file](https://github.com/avicarmeli/rtl_433/blob/master/src/devices/arad_ms_meter.c) you might want to change line 48 to shorter preamble or to one which comply to your 's meter serial number.

Mindful, 2023-07-26:
@avicarmeli This makes sense - ultimately if I can read consumption that will be enough for me as well, so I think I'm just going to have to buy a dongle and play around with it. Thanks for your work on this!

I'll let you know how it goes, although that will probably be at least a few weeks out.

bemasher, 2023-08-13:
See #284

closed 2023-08-13 by bemasher