bemasher/rtlamr #248: 100G DLS - Assuming Encrytped

loca5790, 2022-04-19:
It appears everyone in the area has a 100G DLS I setup RTL AMR and listend for quiet some time and had no messages.  I'm assuming that must mean it's encrypted and it won't pick anything up.  I tuned my tuner in and watched and there is a bunch of chatter unfortunately it lines up with z-wave and I have many z-wave devices.

Any ideas on if picking up usage from encryption is even viable?

loca5790, 2022-04-19:
so I analyzed and found data on frequency 926.41 at 2359300 Hz just going to turn gas on and watch to see for changes.  The pulses over a couple minute period were very repetitive.

loca5790, 2022-04-21:
Found a few meters but in 30 hour period I haven't gotten a duplicate of the same ID.  If I scan the frequency I get pretty consistent tags in RTL_433.  Sensitivity issue?

bemasher, 2022-04-27:
It sounds like you don't have any ERT-compatible meters in your area. It's common to see some spurious messages over the span of a few hours, eventually random noise will look like a valid packet.

There are two likely reasons you won't hear from your meter: it's configured for wake-up instead of bubble-up, meaning it doesn't regularly transmit unless asked to; and second, it doesn't have an ERT-compatible transceiver despite being a meter that usually has one.

loca5790, 2022-04-27:
> It sounds like you don't have any ERT-compatible meters in your area. It's common to see some spurious messages over the span of a few hours, eventually random noise will look like a valid packet. [...]

I was able to get a packet one time from my meter in a 30 hour period.  It matched my consumption displayed but that was the only time I got it.  I'm assuming this must mean it is setup for wake-up.  Anyway to ping it now that I know the address to poll?

bemasher, 2022-05-10:
See: https://github.com/bemasher/rtlamr/issues/103 Send wake-up command?

closed 2022-05-10 by bemasher