bemasher/rtlamr #251: r900: How do you know which water meter is yours?

cypherstream, 2022-04-28:
I have rtlamr working successfully pulling in a few SCM messages which I think are gas meters, but the bul;k of the traffic are R900 which should be the water meters.

I have a neptune e-coder 5/8" T-10 water meter.  On it doesn't say much but it has a small inscription :
3026
0318
Next to a black plastic piece that has a cable going to a box attached to a pipe with an antenna on it.

When I log into our water company website, it says BP#1103126784 and below it has an account number starting with 21003... and below that Meters S/N: 64013190.

However when running rtlamr nothing matches 64013190 or 3026 0318.  I see a lot of these:

{Time:2022-04-28T15:49:18.859 R900:{ID:1481194156 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:11184810 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:49:20.154 R900:{ID:1481302196 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:  201013 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:49:28.687 SCM:{ID:12734750 Type:12 Tamper:{Phy:00 Enc:00} Consumption:  470208 CRC:0x2567}}
{Time:2022-04-28T15:49:38.706 R900:{ID:1550669416 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 1454022 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:49:42.374 R900:{ID:1481194156 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:11184810 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:49:47.156 R900:{ID:1550256522 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 2351496 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:50:01.151 R900:{ID:1550256522 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 2351542 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:50:10.149 R900:{ID:1481302196 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:  201013 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:50:10.371 R900:{ID:1810219234 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 5734946 Unkn3:0x00 Leak: 3 LeakNow:0}}
{Time:2022-04-28T15:50:15.151 R900:{ID:1550256522 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 2351542 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:50:27.352 R900:{ID:1810258482 Unkn1:0xA3 NoUse: 2 BackFlow:1 Consumption: 4501441 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:50:28.185 SCM:{ID:12734750 Type:12 Tamper:{Phy:00 Enc:00} Consumption:  470208 CRC:0x2567}}
{Time:2022-04-28T15:50:29.462 R900:{ID:1481194156 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:11184810 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:50:34.738 R900:{ID:1550669416 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 1454022 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:50:37.033 SCM:{ID: 2484328 Type:12 Tamper:{Phy:00 Enc:00} Consumption:  309422 CRC:0xA2E6}}
{Time:2022-04-28T15:50:58.183 SCM:{ID:12734750 Type:12 Tamper:{Phy:00 Enc:00} Consumption:  470208 CRC:0x2567}}
{Time:2022-04-28T15:51:00.144 R900:{ID:1481302196 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:  201013 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:51:02.695 R900:{ID:1550669416 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 1454022 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:51:06.369 R900:{ID:1810219234 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 5734946 Unkn3:0x00 Leak: 3 LeakNow:0}}
{Time:2022-04-28T15:51:11.151 R900:{ID:1550256522 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 2351542 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:51:16.520 R900:{ID:1481194156 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:11184810 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:51:21.966 SCM:{ID: 1529238 Type:12 Tamper:{Phy:03 Enc:00} Consumption:  142534 CRC:0x5E84}}
{Time:2022-04-28T15:51:23.351 R900:{ID:1810258482 Unkn1:0xA3 NoUse: 2 BackFlow:1 Consumption: 4501441 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:51:25.143 R900:{ID:1481302196 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:  201013 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:51:34.369 R900:{ID:1810219234 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 5734946 Unkn3:0x00 Leak: 3 LeakNow:0}}
{Time:2022-04-28T15:51:39.189 R900:{ID:1550256522 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 2351542 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:51:43.182 SCM:{ID:12734750 Type:12 Tamper:{Phy:00 Enc:00} Consumption:  470208 CRC:0x2567}}
{Time:2022-04-28T15:51:48.576 R900:{ID:1565539334 Unkn1:0xA3 NoUse: 1 BackFlow:0 Consumption:  637004 Unkn3:0x00 Leak: 1 LeakNow:0}}
{Time:2022-04-28T15:51:50.140 R900:{ID:1481302196 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:  201013 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:51:58.738 R900:{ID:1550669416 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 1454022 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:51:59.181 SCM:{ID:12734750 Type:12 Tamper:{Phy:00 Enc:00} Consumption:  470208 CRC:0x2567}}
{Time:2022-04-28T15:52:07.183 R900:{ID:1550256522 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 2351542 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:52:15.138 R900:{ID:1481302196 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:  201013 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:52:18.200 R900:{ID:1810246546 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 6163434 Unkn3:0x00 Leak: 1 LeakNow:0}}
{Time:2022-04-28T15:52:19.349 R900:{ID:1810258482 Unkn1:0xA3 NoUse: 2 BackFlow:1 Consumption: 4501441 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:52:27.091 R900:{ID:1481194156 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:11184810 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:52:30.089 R900:{ID:1481316214 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:   17184 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:52:30.368 R900:{ID:1810219234 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 5734946 Unkn3:0x00 Leak: 3 LeakNow:0}}
{Time:2022-04-28T15:52:35.188 R900:{ID:1550256522 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 2351542 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:52:37.396 SCM:{ID:45501049 Type:12 Tamper:{Phy:02 Enc:00} Consumption:  773992 CRC:0xDE95}}
{Time:2022-04-28T15:52:43.622 SCM:{ID:48952096 Type:12 Tamper:{Phy:00 Enc:00} Consumption:  718768 CRC:0x26B0}}
{Time:2022-04-28T15:52:44.566 R900:{ID:1565539334 Unkn1:0xA3 NoUse: 1 BackFlow:0 Consumption:  637004 Unkn3:0x00 Leak: 1 LeakNow:0}}
{Time:2022-04-28T15:52:47.348 R900:{ID:1810258482 Unkn1:0xA3 NoUse: 2 BackFlow:1 Consumption: 4501441 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:52:54.689 R900:{ID:1550669416 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 1454022 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:53:05.133 R900:{ID:1481302196 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:  201013 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:53:14.177 R900:{ID:1481194156 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:11184810 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:53:21.963 SCM:{ID: 1529238 Type:12 Tamper:{Phy:03 Enc:00} Consumption:  142534 CRC:0x5E84}}
{Time:2022-04-28T15:53:22.732 R900:{ID:1550669416 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 1454022 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:53:31.177 R900:{ID:1550256522 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 2351542 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:53:36.526 SCM:{ID: 2484328 Type:12 Tamper:{Phy:00 Enc:00} Consumption:  309422 CRC:0xA2E6}}
{Time:2022-04-28T15:53:43.347 R900:{ID:1810258482 Unkn1:0xA3 NoUse: 2 BackFlow:1 Consumption: 4501441 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:53:54.396 R900:{ID:1810219234 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 5734946 Unkn3:0x00 Leak: 3 LeakNow:0}}
{Time:2022-04-28T15:53:55.129 R900:{ID:1481302196 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:  201013 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:53:59.187 R900:{ID:1550256522 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 2351542 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:54:01.234 R900:{ID:1481194156 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:11184810 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:54:06.576 SCM:{ID: 2544483 Type:12 Tamper:{Phy:00 Enc:00} Consumption:  240682 CRC:0x8F75}}
{Time:2022-04-28T15:54:18.732 R900:{ID:1550669416 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 1454022 Unkn3:0x00 Leak: 0 LeakNow:0}}
{Time:2022-04-28T15:54:20.127 R900:{ID:1481302196 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:  201013 Unkn3:0x00 Leak:15 LeakNow:3}}
{Time:2022-04-28T15:54:27.172 R900:{ID:1550256522 Unkn1:0xA3 NoUse:32 BackFlow:0 Consumption: 2351542 Unkn3:0x00 Leak: 0 LeakNow:0}}
[... 88 lines cut ...]
{Time:2022-04-28T16:03:14.382 R900:{ID:1810219234 Unkn1:0xA3 NoUse: 0 BackFlow:0 Consumption: 5734946 Unkn3:0x00 Leak: 3 LeakNow:0}}
[... cut ...]
{Time:2022-04-28T16:05:03.712 R900:{ID:1481194156 Unkn1:0xA1 NoUse: 0 BackFlow:0 Consumption:11184810 Unkn3:0x00 Leak:15 LeakNow:3}}

How would I know which one is mine, none of the ID numbers seem to match up?  Our community thinks our water company is corrupt.  For a family of 4 its $233 next month.  The bill average is about $180-$200 usually.  Its extremely hard for some people to afford.  Some customers on our local community sites are complaining of even higher water bills, in the $300 a month or more range.  We are trying to expose this corrupt utility.

Thanks!

doctorkb, 2022-04-28:
The easiest way to verify this (the first time) is to compare the readings -- look for one where the Consumption seems to match yours.  You may note that your meter says something like 14540 and this could match the reading you're seeing of 1454022 -- precision isn't always the same.

cypherstream, 2022-04-28:
Ok so for the water meter theres a solar strip on it, and in January when I had the water company come out and inspect everything due to high bills, he showed me you can shine a flashlight on it and the display comes up.  Keep the flashlight shining on it and the LCD segment display will cycle through different things like the time, consumption reading, leak detection, current water flow rate (which should be 0 if all water is off).

I was in there with my iPhone flashlight and the LCD just doesn't come on.  Is there something different with the iPhone flashlight vs the regular incandescent flashlight the water meter tech had?

Is the ID: XXXXXXXXX  the serial number?  I went online and my Natural Gas company lists my meter number as 1204282, but I don't see that anywhere in the SCM / SCM+ outputs either.  I know that meter says 40G on it (When I got it in 2011 after getting rid of Oil, I made a joke that it had a 40gb hard drive inside).

soundynewind, 2023-05-04:
The meter interface unit (MIU), which is the wired box with the antenna, has its own serial number separate from that of the encoder register.

Check the MIU's FCCID label for a 10-digit SN starting with 1. If the label is missing/illegible, then ask your utility for the info (I'd recommend contact via email/feedback form submission since first-tier call center staff may not understand, or even have access to, what you want). If the utility won't comply, then there are some alternate methods:
- draw a lot of water in the middle of the night and then match against the ID transmitting rapidly-increasing consumption values over time
- intentionally attenuate the signal into your SDR (e.g. filter or remove antenna) and then place SDR near the MIU's antenna

closed 2023-08-13 by bemasher