bemasher/rtlamr #258: Support meter type 19?

PinkFreud, 2022-07-10:
When I had solar installed here, my utility company (PSE&G, NJ) wound up installing a net meter.  The ID label seems to show a meter type of 19, which I have yet to see anywhere else.  Unsurprisingly, my meter ID fails to show up in rtlamr.
[image]

I have a suspicion that type 19 may be similar to type 12, though, for two particular reasons:
1. PSE&G handles both electricity and gas for me.  My gas meter is a type 12, and shows up in rtlamr.  It seems unlikely that PSE&G would have two vastly different technologies for reading meters, as it would increase the complexity of remote reading.
2. As evidenced by [an image on an old blog post](http://4.bp.blogspot.com/-PQzxMG4h_f0/UA7PxalB0pI/AAAAAAAAAHo/Dg427yzqgWc/s1600/0724021058.jpg) about solar from 10 years ago, PSE&G was installing type 12 net meters a decade ago.  Sometime between then and now they appear to have switched to type 19.

Would it be possible to add support for type 19 meters?  I'm pretty new to SDR in general, so if there's a way I can investigate output of rtlsdr for any mention of my meter id, I'd be happy to see what I can dig up.

bemasher, 2022-07-13:
It's pretty likely your meter is already transmitting ERT-flavored messages, it may just be a matter of finding the right message type, and frequency the meter is transmitting on. I'd recommend trying SCM, SCM+, IDM, and NETIDM. You may also want to write a simple script to launch and dwell on a series of frequencies using the `-centerfreq` and `-duration` flags.

PinkFreud, 2022-07-13:
I've tried -msgtype=all, but wasn't able to find this meter's ID in the output.  Of course, things would be slightly simpler if I could find the standard FCC label on this meter - I'll take another look to see if it's somewhere visible but less obvious, otherwise it's likely hidden inside the locked enclosure somewhere.

I'll play with frequencies when I get a spare few minutes to see if I can discover where this thing is hiding.  I also find myself wondering if the reason for this new meter type might be encryption, in which case, I might be SOL.

PinkFreud, 2022-07-13:
Ouch - just went digging through Xylem/Sensus' marketing materials on the Aclara I-210+c.  It looks like this meter can be equipped with one of two different modules:

```
DT-20

    Under cover, fully rated 200 amp connect/disconnect switch
    Daily self reads
    kWh and optional kVArh and kVAh energy measurements with associated demand values
    ToU flexibility with four tiers, four seasons, 50 perpetual holidays and critical piece pricing (CPP)
    Enhanced performance with four channels of ANSI C12.19 load profile data and up to 14 snapshots

DT-105

    Under cover, fully rated 200 amp connect/disconnect switch
    Daily self reads
    kWh and optional kVArh and kVAh energy measurements with associated demand values
    ToU flexibility with four tiers, four seasons, 50 perpetual holidays and critical piece pricing (CPP)
    Enhanced performance with four channels of ANSI C12.19 load profile data and up to 15 snapshots
    Time and date stamped event log
    Full meter programming and firmware updates over the air
    Zigbee equipped
    Voltage sag and swell log
```
[Source](https://www.xylem.com/en-us/products--services/metrology-equipment-for-utilities/meters/aclara-i-210c-electricity-meter/)

They're also heavily pushing a proprietary product they're calling FlexNet, which they're marketing as AMI (vs AMR).  It's difficult to tell from the marketing materials, but I suspect that's the purpose of the ZigBee-equipped DT-105 module.

They also appear to [sell](https://sensus.com/solutions/ert-meter-reading/) an all-in-one solution to read both older AMR meters and newer AMI meters, which I suspect is what my utility company is using.  Thus, the difference between type 19 and 12 may well be use of Xylem's AMI infrastructure, which means this is going to get very interesting indeed.

dmkjr, 2022-08-16:
@PinkFreud I just received a new meter and the literature shows it's AMI.  I'm assuming this means we aren't going to be able to pick it up on our 900mhz antenna's anymore?

bemasher, 2023-08-13:
See #284

closed 2023-08-13 by bemasher