bemasher/rtlamr #262: Tried in the UK on an Elster AS300P electricity meter

nitropaul, 2022-10-11:
I have installed the 'rtlamr2mqtt' addon into HomeAssistant, i have connected my SDR device and tried scanning for my electricity / gas meter but I dont see anything. I know they are not in the supported devices but i thought I would give it a go. My config is standard as it was installed and my log file is as below. Does it look like it is scanning properly? Is it probably just that my meter uses a different protocol?
[2022-10-11 18:08:19] Using "/data/options.json" config file
[2022-10-11 18:08:19] Fetching default MQTT configuration from http://supervisor/services/mqtt
[2022-10-11 18:08:19] RTLAMR2MQTT Starting...
[2022-10-11 18:08:19] RTL SDR Device 0bda:2838 found on USB port 001:005 - Index: 0
[2022-10-11 18:08:19] No USB device specified in the config file, using the first found.
[2022-10-11 18:08:19] Configured MQTT sender:
[2022-10-11 18:08:19]  > hostname => core-mosquitto
[2022-10-11 18:08:19]  > port => 1883
[2022-10-11 18:08:19]  > username => addons
[2022-10-11 18:08:19]  > client_id => rtlamr2mqtt
[2022-10-11 18:08:19]  > base_topic => rtlamr
[2022-10-11 18:08:19]  > availability_topic => rtlamr/status
[2022-10-11 18:08:19]  > tls => None
[2022-10-11 18:08:19] Reseting USB device: /dev/bus/usb/001/005
[2022-10-11 18:08:19] Reset sucessful.
[2022-10-11 18:08:19] Sending message to MQTT:
[2022-10-11 18:08:19]  > topic => rtlamr/status
[2022-10-11 18:08:19]  > payload => online
[2022-10-11 18:08:19]  > retain => True
[2022-10-11 18:08:19] Trying to start RTL_TCP: /usr/bin/rtl_tcp  -s 2048000
[2022-10-11 18:08:19] RTL_TCP started with PID 10
[2022-10-11 18:08:24] RTL_TCP is ready to receive connections!
[2022-10-11 18:08:24] Trying to start RTLAMR: /usr/bin/rtlamr -msgtype=scm+ -format=json -filterid=123456789 -unique=true
[2022-10-11 18:08:24] RTLAMR started with PID 15
[2022-10-11 18:08:25] 18:08:25.125512 decode.go:45: CenterFreq: 912600155
[2022-10-11 18:08:25] 18:08:25.126348 decode.go:46: SampleRate: 2359296
[2022-10-11 18:08:25] 18:08:25.126388 decode.go:47: DataRate: 32768
[2022-10-11 18:08:25] 18:08:25.126412 decode.go:48: ChipLength: 72
[2022-10-11 18:08:25] 18:08:25.126434 decode.go:49: PreambleSymbols: 16
[2022-10-11 18:08:25] 18:08:25.126456 decode.go:50: PreambleLength: 2304
[2022-10-11 18:08:25] 18:08:25.126476 decode.go:51: PacketSymbols: 128
[2022-10-11 18:08:25] 18:08:25.126498 decode.go:52: PacketLength: 18432
[2022-10-11 18:08:25] 18:08:25.126525 decode.go:59: Protocols: scm+
[2022-10-11 18:08:25] 18:08:25.126547 decode.go:60: Preambles: 0001011010100011
[2022-10-11 18:08:25] 18:08:25.126570 main.go:124: GainCount: 29

nitropaul, 2022-10-11:
If I set listen_only to 'true' i get the following log:
[2022-10-11 18:23:57] Using "/data/options.json" config file
[2022-10-11 18:23:57] Fetching default MQTT configuration from http://supervisor/services/mqtt
[2022-10-11 18:23:57] Starting in LISTEN ONLY Mode...
[2022-10-11 18:23:57] Using "/data/options.json" config file
[2022-10-11 18:23:57] Fetching default MQTT configuration from http://supervisor/services/mqtt
[2022-10-11 18:23:57] Configured MQTT sender:
[2022-10-11 18:23:57]  > hostname => core-mosquitto
[2022-10-11 18:23:57]  > port => 1883
[2022-10-11 18:23:57]  > username => addons
[2022-10-11 18:23:57]  > client_id => rtlamr2mqtt
[2022-10-11 18:23:57]  > base_topic => rtlamr
[2022-10-11 18:23:57]  > availability_topic => rtlamr/status
[2022-10-11 18:23:57]  > tls => None
[2022-10-11 18:23:57] Starting rtl_tcp with /usr/bin/rtl_tcp 
[2022-10-11 18:23:59] Starting rtlamr with ['/usr/bin/rtlamr', '-msgtype=all', '-format=json']
[2022-10-11 18:23:59] You should see all utility meters after this line:
[2022-10-11 18:24:00] 18:24:00.202956 decode.go:45: CenterFreq: 912600155
[2022-10-11 18:24:00] 18:24:00.203773 decode.go:46: SampleRate: 2359296
[2022-10-11 18:24:00] 18:24:00.203982 decode.go:47: DataRate: 32768
[2022-10-11 18:24:00] 18:24:00.204184 decode.go:48: ChipLength: 72
[2022-10-11 18:24:00] 18:24:00.204363 decode.go:49: PreambleSymbols: 32
[2022-10-11 18:24:00] 18:24:00.204543 decode.go:50: PreambleLength: 4608
[2022-10-11 18:24:00] 18:24:00.204758 decode.go:51: PacketSymbols: 736
[2022-10-11 18:24:00] 18:24:00.204956 decode.go:52: PacketLength: 105984
[2022-10-11 18:24:00] 18:24:00.205176 decode.go:59: Protocols: idm,r900,scm,scm+
[2022-10-11 18:24:00] 18:24:00.205397 decode.go:60: Preambles: 01010101010101010001011010100011,00000000000000001110010101100100,111110010101001100000,0001011010100011
[2022-10-11 18:24:00] 18:24:00.205647 main.go:124: GainCount: 29

bemasher, 2022-10-12:
Both of those logs look normal for rtlamr, but probably there are no supported meters in range.

closed 2022-10-12 by bemasher