bemasher/rtlamr #266: Integrating American Meter Company model AC-250 Gas Meter

moasson, 2022-11-04:
Hi all,

I've been using the [rtlamr2mqtt](https://github.com/allangood/rtlamr2mqtt) project to monitor my electricity. It's working well. This led me to try to integrate my gas meter consumption into HA as well, however, I am unable to locate my meter when running the add-on in docker using the DEBUG MODE.

My gas meter is model AC-250 from American Meter Company. Hope someone here can assist me on making this work if it possible.

[image]

moasson, 2022-11-04:
Ah, ok. So, I should still be able to pick it up if I keep the scan running with the meter ID? What would be the meter ID here? Is it the number on the meter itself or the meter number showing up in my gas bill?

bemasher, 2023-08-13:
See #284.

closed 2023-08-13 by bemasher

tismofied, 2026-11-13:
@amdjml I am sorry to wake up this post from it's sleep but I have the same exact meter. Did you manage to get it going and if you can remember what you did that would be very helpful and appreciated.