bemasher/rtlamr #267: RTL_TCP quits

chalupien, 2022-11-10:
Hi Im running two separate windows cmd prompts. One running RTL_TCP and ONE Running RTLAMR | RTLAMR-COLLECT.

It appears RTL_TCP quits unexpectedly and I have to restart both. The timing does not match to anything in particular. I have read that the quality of the SDR dongle can help with this "crash". Just wanted to get some more info on what others have done to remedy this issue. 

I have also heard of others writing a "watchdog" program to determine a crash. Any help would be great.

Thanks

Nils154, 2022-11-10:
Used to happen frequently on my pi2.
Also happened on my pi4, but only on the usb2.0 ports.

Once I put my dongle on the usb3.0 port, I haven’t had any issues.

chalupien, 2022-11-12:
Seems to be working without quitting. Any idea how to monitor the rtl_tcp service and restart if it quits?

chalupien, 2022-11-13:
Well just stopped. So that did not help. Any other ideas?

XjSv, 2022-12-29:
I'm having the same issue. When I run `rtlamr` with `--single=true`, the first time it runs fine and returns the json. Although I do see this message:

`21:05:34.448953 main.go:343: Receiver context cancelled.`

But on the second run `rtl_tcp` crashes. 

Command: `rtlamr -msgtype=scm --format=json --filterid=00000000 --single=true`

Log
```
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Tuned to 100000000 Hz.
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source <---- * 1st RUN *
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
client accepted!
set freq 912600155
set sample rate 2359296
set gain mode 0
comm recv bye
Signal caught, exiting!
worker cond timeout
Signal caught, exiting!
all threads dead..
listening...
Use the device argument 'rtl_tcp=127.0.0.1:1234' in OsmoSDR (gr-osmosdr) source <---- * 2nd RUN *
to receive samples in GRC and control rtl_tcp parameters (frequency, gain, ...).
client accepted!
set freq 912600155
<---- * CRASH *
```

While testing I've also come across this message:

```
client accepted!
set freq 9c1b2 6t0r0a1n5s5f
er status: 1, canceling...
<---- * CRASH *
```

I just started using rtlamr and rtl_tcp, so it's possible that I'm doing something wrong.

infecticide, 2024-01-25:
I run my rtl_tcp with a systemd unit:

`[Unit]
Description=RTL-SDR TCP Server
Wants=network-online.target
After=network-online.target

[Service]
ExecStartPre=-/bin/echo 0 > /sys/module/usbcore/parameters/usbfs_memory_mb
ExecStartPre=/bin/sleep 5
ExecStart=/usr/bin/rtl_tcp -g 0 -d 0
WorkingDirectory=/home/infecticide
StandardOutput=inherit
StandardError=inherit
Restart=always

[Install]
WantedBy=multi-user.target
`

closed 2025-08-15 by bemasher