bemasher/rtlamr #276: r900Parser.filter / quantize takes over 80% of CPU

jdevelop, 2023-05-08:
I noticed that rtl_amr app uses over 85% of CPU on my NUC box ( core i7 ) also tried on my laptop with Xeon, same results.

Tried with most recent "master".

With help of `pprof` I narrowed it down to

```
      flat  flat%   sum%        cum   cum%
    6140ms 37.81% 37.81%     6140ms 37.81%  github.com/bemasher/rtlamr/r900.Parser.filter
    5850ms 36.02% 73.83%     6640ms 40.89%  github.com/bemasher/rtlamr/r900.Parser.quantize
     630ms  3.88% 77.71%      630ms  3.88%  github.com/bemasher/rtlamr/protocol.Decoder.Filter
     470ms  2.89% 80.60%      470ms  2.89%  github.com/bemasher/rtlamr/protocol.MagLUT.Execute
     410ms  2.52% 83.13%      790ms  4.86%  math.Abs (inline)
     400ms  2.46% 85.59%      730ms  4.50%  github.com/bemasher/rtlamr/protocol.(*Decoder).Search
     400ms  2.46% 88.05%      400ms  2.46%  runtime.memmove
     330ms  2.03% 90.09%      330ms  2.03%  github.com/bemasher/rtlamr/protocol.searchPassByte (inline)
     250ms  1.54% 91.63%      250ms  1.54%  math.Float64frombits (inline)
     200ms  1.23% 92.86%      200ms  1.23%  runtime.futex
```

I can see that this is rather pure math stuff going on in there ( and quite not trivial ), but just in case reporting this as someone can take a look and perhaps come up with some optimizations.

bemasher, 2023-05-08:
Unfortunately this demodulator is a lot more involved than the manchester coding used in the rest of the message types.

Probably it could be optimized, but it's not a device I own so I don't have much motivation to spend more time on it than I already have. If someone else would like to take a crack at shoring it up, please do.

closed 2023-08-13 by bemasher

jdevelop, 2024-01-31:
@bemasher hi,I got some spare time and I'd like to see if I can optimize the R900 parser code. Any chance you could share some protocol definitions? I googled for that R900 thing and in so far everything I find is a bit vague.

bemasher, 2024-01-31:
The current implementation uses matched filters, one for each symbol pair (symbol and it's inversion). Then quantizes the output of those matched filters into the symbol with the best match at each sample in the signal.

These are the symbols:
https://github.com/bemasher/rtlamr/blob/dcdddc5a6e7717a038e9346d0adcf9669e4f60b7/r900/r900.go#L132-L134