bemasher/rtlamr #280: panic on startup

utdrmac, 2023-08-03:
Made a custom docker container with rtl-sdr, rtlamr, and rtlamr-collect. When it starts up, I think there's a race condition that I'm working out where something starts before something else resulting in this panic.

```
00:59:39.771085 main.go:90: Error connecting to spectrum server: dial tcp 127.0.0.1:1234: connect: connection refused
rcvr.Connect
main.(*Receiver).NewReceiver
	/go/pkg/mod/github.com/bemasher/rtlamr@v0.9.3/main.go:90
main.main
	/go/pkg/mod/github.com/bemasher/rtlamr@v0.9.3/main.go:313
runtime.main
	/usr/local/go/src/runtime/proc.go:250
runtime.goexit
	/usr/local/go/src/runtime/asm_amd64.s:1598
```

bemasher, 2023-08-03:
Yep, you'll need to be sure the spectrum server is running and reachable before rtlamr starts.

If the spectrum server is in another container, I don't think you'll be able to reach it at a localhost anyway.

utdrmac, 2023-08-03:
Everything is in 1 container. I submitted this bug report as I figured you'd want to trap the panic and do some sleep-retry logic for a better user experience.

bemasher, 2023-08-05:
I agree, it would be a quality of life improvement. I'll have a look at it.

Oxalin, 2023-11-22:
I'm not using a container and I'm hitting the same error under Raspberry Pi OS Bookworm. Do you think it is related?

Oxalin, 2023-11-22:
> I'm not using a container and I'm hitting the same error under Raspberry Pi OS Bookworm. Do you think it is related?

Forget about this, my problem is unrelated. Calling 127.0.0.1 fails, but not using the "external" IP. This is a network configuration problem on my Pi.

closed 2025-08-15 by bemasher