bemasher/rtlamr #285: R900 meters may need additional decoding?

pyrodex, 2023-09-07:
I recently found this project and love it for the simplistic nature!

Before this project I was running rtl_433 (nightly) and was getting data from my Gas (ERT-SCM) and Water (R900) meters without issues. When I switched over to this project I loved the fact I could ignore the noise from the neighbors and only collect my meters. With that being said I have noticed a possible decoding issue between rtl_433 project and this effort. 

This is all my data so I am not sharing anyone else in my direct area.

My water meter as rtl_433 saw it:

```
{
    "time": "2023-09-07T14:48:59.974631-0400",
    "protocol": 228,
    "model": "Neptune-R900",
    "id": 701300298,
    "unkn1": 131,
    "unkn2": 4,
    "nouse": 0,
    "backflow": 0,
    "consumption": 1054604,
    "unkn3": 0,
    "leak": 0,
    "leaknow": 0,
    "extra": "203105",
    "mod": "ASK",
    "freq": 915.01626,
    "rssi": -0.139523,
    "snr": 33.55369,
    "noise": -33.6932
}
```

After switching to rtlamr I see the following data:

```
{
    "Time": "2023-09-07T20:28:19.962017499Z",
    "Offset": 0,
    "Length": 0,
    "Type": "R900",
    "Message": {
        "ID": 701300298,
        "Unkn1": 131,
        "NoUse": 32,
        "BackFlow": 0,
        "Consumption": 1054640,
        "Unkn3": 0,
        "Leak": 0,
        "LeakNow": 0
    }
}
```

Notice how the nouse/NoUse is set to 0 and also there is the missing unkn2 field? I am not sure if there is an issue in the decoding structure of either project and just trying to make sure everything is the same if possible. 

My reference is of course from https://github.com/merbanan/rtl_433/blob/master/src/devices/neptune_r900.c which clearly references this project so not sure who can help :).

charlesfayal, 2023-10-30:
Curious what you mean by you could ignore the noise from your neighbors?

richiecrews1980, 2023-10-30:
> Curious what you mean by you could ignore the noise from your neighbors?

I liked rtlamr would pull JUST the IDs I wanted and with a solution like rtl_433 its a flood and I can't narrow down by device ID but instead narrow down by device.

pyrodex, 2023-10-30:
> Curious what you mean by you could ignore the noise from your neighbors?

I liked rtlamr would pull JUST the IDs I wanted and with a solution like rtl_433 its a flood and I can't narrow down by device ID but instead narrow down by device.

closed 2025-08-15 by bemasher