bemasher/rtlamr #292: Itron 500G stopped working

m0sia, 2024-02-05:
Hi,

I have a Itron 500G gas meter and it was working fine for more than a year. Suddenly at 2am in the morning of November 2, 2023 I stopped receiving any reading from it as well as any of the neighbors' readings.

At first I was blaming the SDR dongle and even bought a new one, but it did not help. Does anyone have an idea what happen? I haven't changed anything with my setup.

chelming, 2024-02-26:
what was your setup for reading it? I wonder if it got switch to only broadcast on demand?

m0sia, 2024-02-26:
Frankly nothing unique. 

I have RTL2832U + R820T2 USB dongle. Initially I started with rtlamr on my MacBook figuring out what I can receive and then I had it running with rtl_433 docker container dumping sensor readings into MQTT/HA.

After it stopped working I  tried the following: 

* Bought a brand new SDR dongle
* Tried both latest rtlamr and rtl_433 both on Linux and MacOS in different locations across the house
* Tried running both on different central frequencies with 1MHz step from 902MHz to 928MHz
* Checked spectrum analyzer if there is any interference and there is none
* Checked spectrum analyzer and I can see some signals in 900MHz band that are not decoded. Not sure if they are from the gas(Itron 500G), electricity(Itron I-210+C) or water(Badger Read-o-Matic) meter

cefoster0, 2025-02-28:
What as your center freq prior to it stopping? I just got a 500g installed and I don't see anything other than water meters.

cefoster0, 2026-03-28:
> [@m0sia](https://github.com/m0sia) [@cefoster0](https://github.com/cefoster0) - curious if you learned anything else on this. I just had a 500G installed today and can't seem to receive anything from it using rtlamr, whereas I used to be able to with the 100G that I had before.

I listened for weeks and never heard another gas message, just my water meter pinning incessantly with nonsense. 

Ended up going with the magnetometer setup, which has its own issues but at least worked:
https://github.com/tronikos/esphome-magnetometer-water-gas-meter

baf, 2026-03-28:
Appreciate the reply. I use that same setup for my water meter since that was changed out for a cellular model last year, but might be harder for my gas meter, which is outdoors. I’ll put some thought into it. Thanks!