bemasher/rtlamr #297: can't read itron cl200 c1sr  (fcc sk9c1a-3b)

fjleon1980, 2024-04-06:
thank you for this software. i can't seem to get a read on this electric type 7 meter itron cl200 c1sr  (fcc sk9c1a-3b)

since you mention c1sr should work i'm wondering if not all c1sr are equal.

doing some searching i found "Operates using spread spectrum technology" so i wonder if it's a different protocol altogether.

this reddit post https://www.reddit.com/r/RTLSDR/comments/10fsj93/reading_itron_ert_scm_data_help although not for the same exact model suggests it may be transmitting using cellnet technology so I may be experiencing the same issue.

FCC https://fcc.io/SK9C1A-3B says this operates within  909.6 - 921.8 mhz. Is there anything I can do to confirm whether I am using rtlamr correctly? I am not passing any parameters to rtl_tcp or rtlamr

fjleon1980, 2024-04-06:
perhaps there is some hope. i used rtl_433 -f 912M and just saw my meter. I tried this yesterday but no luck. maybe it's not transmitting frequently?

model     : SCMplus
id        : <omitted>
Protocol_ID: 0x1E
Endpoint_Type: 0x07
Endpoint_ID: <omitted>
Consumption: 435272 (yesterday i took a picture of my meter and it said 04253, maybe right now it would say 04352?)
Tamper    : 0x0301
crc       : 0x63D2
Meter_Type: Electric
Integrity : CRC

model     : IDM
PacketTypeID: 0x1C
PacketLength: 92
ApplicationVersion: 4
ERTType   : 0x17
ERTSerialNumber: <omitted>
ConsumptionIntervalCount: 194
ModuleProgrammingState: 0xBC
TamperCounters: 0x030100070D00
AsynchronousCounters: 0x00
PowerOutageFlags: 0x000000000000
LastConsumptionCount: 435272
DifferentialConsumptionIntervals: 3, 3, 4, 3, 6, 6, 7, 6, 6, 5, 7, 19, 23, 4, 5, 6, 6, 6, 5, 6, 3, 2, 4, 2, 4, 2, 4, 2,>TransmitTimeOffset: 3432
MeterIdCRC: 0x9631
PacketCRC : 0xCA63
Meter_Type: Electric
Integrity : CRC

time      : 2024-04-06 11:30:41
model     : NETIDM
PacketTypeID: 0x1C
PacketLength: 92
ApplicationVersion: 4
ERTType   : 0x17
ERTSerialNumber: <omitted>
ConsumptionIntervalCount: 194
ModuleProgrammingState: 0xBC
TamperCounters: 0x030100070D00
Unknown_field_1: 0x00000000000000
LastGenerationCount: 6
Unknown_field_2: 0xA44801
LastConsumptionCount: -2134867920                                                                                       DifferentialConsumptionIntervals: 3078, 224, 6156, 320, 14412, 2945, 40, 1539, 96, 2566, 96, 2056, 128, 8200, 512, 8208>TransmitTimeOffset: 3432
MeterIdCRC: 0x9631
PacketCRC : 0xCA63
MeterType : Electric                    

it transmitted twice between 4 minutes although didn't see any readings at the third set. is there any special parameter i should use with rtlamr or i just got lucky?

jakew02, 2025-07-09:
I know this is old but i have the same meter. I wasn't getting readings, but i just let it sit for a few minutes and it did eventually pick it up.

You will have two barcodes, I used the ID on the *top* barcode and it is working well.