bemasher/rtlamr #299: Itron 100WD

andhab1, 2024-04-22:
There is some indication in other messages here that the Itron 100WD (100W+ ?) is supported. I tried scm and scm+, and netidm but i don't see it what looks like my water meter appear anywhere. I am getting a lot of gas meters which appear to be scm type. 
In another post it was suggested that this meter might actually be nim, and there was some attempt to record with rtlcap. Is this still helpful? I can try to do the same if it helps make progress.
for documentation the, FCC ID looks like EWQ100WD, Also, is my ID actually the 79216200 number?
[image]

bemasher, 2024-04-22:
Looks like it may be commodity type 11, but I'm not sure how to interpret the ERW-1300-403 following.

Captures may be useful here since the label is similar to other ERT-capable meters.

andhab1, 2024-04-22:
Here is a better image with less info cut off. Can you recommend the command line arguments for capture?
[image]

CasaBains, 2024-11-16:
Did you find a way to capture it? I have the same model for my city water meter.

argilo, 2025-11-05:
I have a 100WD meter, and can see its readings as long as I run rtlamr with `-msgtype=scm+`.

dudeofawesome, 2026-10-10:
Same findings as @argilo, though my utility's 100WD / meter is only configured to report integer HCF, so the readings are basically useless for leak detection.