bemasher/rtlamr #303: Documentation clarification request:  Handling spread spectrum

Entropy512, 2024-06-16:
All documentation I've seen on the iTron ERT protocols state that they are FHSS, however I don't see any evidence of handling frequency hopping in rtlamr anywhere (but maybe I haven't fully understood the code yet...)

Is rtlamr:

- Just sitting on one channel and iTron always transmits a full packet before switching channels
- Using enough bandwidth to capture multiple channels and just look for "Is there energy anywhere in the capture bandwidth?" - it looks like an RTL dongle could only capture a few channels at most though
- Something else?  I may be missing it but I don't see any evidence of frequency conversion + filtering at lower bandwidths to simultaneously demodulate individual channels

I'm looking into reading an iTron Gen5 Riva - I originally saw that it was FHSS in the FCC documentation and gave up, but if receiving it is possible by just sitting on one of the hopping channels and waiting, there might be a chance.

I'm not asking for rtlamr to support it - I plan on doing the work myself (especially if "sit on a single channel" is valid because then I can use a CC1101 to demodulate with minimal CPU usage), but I've been using rtlamr as a rough starting point in the case that the Gen5 Riva has only changed modulation schemes and not the higher-level protocols.  (The Gen5 Riva supports multiple FSK and GFSK bitrates, but no OOK.) - but some basic knowledge of how rtlamr is handling the fact that ERT is spread spectrum would be useful.

Entropy512, 2024-06-26:
OK, after doing a lot of digging and experimentation with both rtlamr and rtl_433, it's obvious that both are relying on the ability to sit on 1-2.3 MHz of spectrum and demodulate anything that appears in it.  Although the default of 912.38 MHz is not even remotely centered on the bottom band of 17 channels out of 50.  (There's a gap from 913.189 MHz up to just below 915 MHz)  912.14 would be the center of the band.

[image]

closed 2024-06-26 by Entropy512

shaunhey, 2024-07-11:
@Entropy512 If you have any information about the Gen5, I'd love to see it. Our city just switched from Elster to Itron Gen5, so I'm looking to understand this new protocol. I had previously developed SDR and rfm69 based solutions for the Elster setup, and look to do the same for Itron Gen5 if it isn't supported in rtlamr.