bemasher/rtlamr #305: Interference from 900MHz (?) Neptune R900 water meter cellular transponder

puterboy, 2024-07-25:
I have been using rltamr to read my electric meter for many months now without issue -- the meter transmits a message every couple of seconds.

About a month ago, I stopped receiving regular messages from the electric meter. Sometimes no messages would be sent for a day or more. When messages were sent, it was mostly for a couple of hours at night.

At first, I thought there was an issue with my electric meter signal transmission. Maybe the transmitter was getting flakey? Maybe it relied on a battery that was running low (though that seemed unlikely for an electric meter). Maybe the incredibly hot weather was causing the transmitter to fail during the hot days but worked a little during the cooler nights?

Then I remembered that our water company installed a new meter and transmitter. In particular, they moved from a local transponder to an LTE cellular one -- specifically a Neptune R900 -- which happens to operate in the 910-920 MHz range.

Unfortunately, the transmitter is only inches away from the electric meter.
Using my laptop with an RTL-SDR dongle and antenna, I verified, that I could only get rtlamr packets) from in front of the meter or on the side opposite the Neptune R900. I couldn't get any signal in my house.

So it seems like the R900 transmission is almost completely blocking the electric meter rtlamr messages.

Given that the R900 uses a battery, I am surprised that it transmits seemingly constantly given that it essentially totally blocks the electric meter rtlamr signals that I know are transmitted about every 2 seconds. Perhaps it's faulty?

Alternatively, I will try to get the water company to move the transmitter further away from the electric meter.

Is there any way though to filter out the R900 signal? 
- Can I set a "narrower" band for rtlamr outside the 910-920 MHz band used by the R900 but that would still be in range of the electric meter transmissions.
- Could I "shield" the R900 device perhaps by wrapping it at least partially in aluminum foil? Would placing a metal plate behind it help to prevent it's signal from entering the house while letting the electric meter signals penetrate freely?

Any other suggestions????

bemasher, 2024-07-26:
It's extremely unlikely that your R900 is transmitting enough that rtlamr can't receive anything from your power meter. I'd fire up an application like SDR# and have a look at the spectrum between 902 and 928 MHz.

puterboy, 2024-08-02:
I am 99% sure it is the problem.
1. Problem happened *exactly* when new R900 cellular transmitter was installed
2. When I probed with my rtl-sdr antenna, I was unable to get *any* electric meter readings on the side of the meter adjacent to the new R900 cellular unit but was able to receive packets on the other side
3. This problem only affects my electric meter and not my water meter or for that matter other neighbor meters that I can still read fine (recall the R900 cellular transmitter is only inches away from my electric meter)

So pretty near certain problem is the R900.

bemasher, 2024-08-03:
One possibility is that the signal from the R900 is high enough power it's overloading the receiver, even though it's not within the sampled bandwidth, try reducing the dongle's gain.

I'd still recommend having a look at the received spectrum with something like SDR# before messing with rtlamr's settings.

puterboy, 2024-08-06:
If I am understanding things correctly, I think this rtl-sdr spectrum analyzer picture explains the problem...
There seems to be a constant ~-44dB SNR signal at 912.040 MHz.
Since the default center frequency for rtlamr is 912.6 MHz, this could explain the interference and why setting the center frequency to 911MHz seems to help a little.

Other than the coincidence of the R900 Neptune device being installed just when I started having trouble grabbing rtlamr signals, I don't know what else has changed around my house that would account for such a strong, new signal.

Any thoughts on what to troubleshoot and/or do beyond moving the center frequency of rtlamr away from 912MHz?

[image]

bemasher, 2024-08-06:
That is the DC offset of the receiver, it will always be present at 0 Hz (relative to the sampled spectrum), you'll find no matter where you tune, it'll follow you around.

In any case, rtlamr is immune to that spike because manchester coding has odd symmetry and it cancels out.

Can you provide a screenshot of the full 2.4 MHz of spectrum?

puterboy, 2024-08-06:
2.4MHz from where to where?

bemasher, 2024-08-06:
Set the receiver to 2.4 MHz sample rate, and tune to 912.6 MHz.

puterboy, 2024-08-06:
OK - meanwhile, btw, here is the spectrum from 911 to 913 MHz -- and you can see a clear spike at 912.04 MHz that is -53dB relative to the background of about -85dB

puterboy, 2024-08-06:
OK here is with 2.4MHz sample rate and tuned to 912.6 MHz

[image]

bemasher, 2024-08-06:
That looks suspiciously empty. What is the gain set to, and is AGC enabled?

Do you have another known-good receiver you can test with?

puterboy, 2024-08-06:
Sorry, I must have taken a bad snapshot.
Here btw is one from another rtl_sdr dongle showing the same 912.04 MHz peak as in comment https://github.com/bemasher/rtlamr/issues/305#issuecomment-2270263681
[image]

bemasher, 2024-08-06:
Still suspiciously empty. I'd expect to see a much wider signal causing interference, or a lot of short bursts all over the spectrum.

What is your gain set to?

puterboy, 2024-08-06:
My gain is set to the default for Airspy-SDR.
OK - maybe the problem is that I was using "Direct Sampling (I Branch)."
When I switch to "Direct Sampling (Q Branch)" I get:
[image]

Is this more like what you would expect?

puterboy, 2024-08-06:
I should also note that the interference is less late at night -- as evidenced by the fact that in the past hour I have been able to pick up my electric meter even with the default center frequency. So the interference is presumably less at this hour.

bemasher, 2024-08-06:
You'll want quadrature sampling and max out the RF Gain slider, should be near 50dB. You're looking for short bursts like this, if you don't see any, I'm not sure what your specific problem is.

[image]
[image]
[image]

puterboy, 2024-08-06:
Ahhh... I had gain set to 0 (default) and I was not using quadrature samplling
Here is what it looks like with Quadrature Sampling and AGC:
[image]

I assume those brief horizontal lines are what you are talking about... and that they correspond to AMR messages

And here it is with gain set to 30:
[image]

And here with gain set to 40:
[image]

But is it normal to have the more prominent, relatively constant noise centered at 912.04???

puterboy, 2024-08-06:
Now here is something really weird..
Correlating the "blackout" periods when I had trouble with reading my electric meter with the temperature, I found that the blackout periods were substantially hotter than the non-blackout periods -- this would explain why I rarely had blackouts during the late night (like last night) except during the most extreme heat spells. Conversely, today it is quite cool out (about low 70's) and the signal has been strong all day.

Some possibilities could include:
1. Neighboring air conditioners are causing interference (however 912 MHz would be a very high and specific harmonic to cause such disruption, plus only my electric meter seems to show such blackout as rtlamr continues to read other meters)
2. The power to the AMR transmitter in my electric meter causes flakiness when the street voltage drops or gets noisy due to high AC load in the neighborhood - but I don't see much correlation between blackout periods and recorded voltage inside in one of my smart meters)
3. The R900 transmitter has some borderline flaky hardware that starts spewing a broader noise pattern in the 900 MHz band when the temperature outside rises causing the temperature in the R900 case to rise -- this doesn't seem that likely

Given the key symptoms:
1. Smothering of rtlamr signal seems directionally and causally related to the R900 (i.e., problem started when the device was installed and the disruption is most severe on the side of the electric meter facing the R900)
2. Disruption seems to be specific to the 912 MHz frequency as dropping the center frequency to 911 prevents drops (I don't seem to have missed anything since I changed the center frequency whereas in the weeks before I would get gaps every day sometimes lasting for up to 2 days)
3. Disruptions seem to be correlated with outside temperature (starting in the late morning as temps warm up and returning in late evening as the air cools off -- or continuing to next day if night is still warm)

This has become quite the mystery for me...

puterboy, 2024-08-07:
I'm thinking more and more that you are right and that it's not the R900 -- indeed, I spoke to the city engineer and the R900 cellular unit doesn't broadcast anything in the 900MHz band but rather via an LTE network.

The strong temperature correlation has me suspecting our neighbors' AC
(indeed with it being cool for the past 36 hours, I have consistently gotten an average of 3 readings per *SECOND* (12000+ plus per hour) from the electric meter vs. no signals for hours at a time during high outside temps.

Will know for sure soon since I got the city to swap out the LTE anyway for a 900MHz packet version -- it works for me since then I can read the packets using your rtlamr code and they get to redeploy the scarce cellular transmitters elsewhere where they need it.

I'll post back to confirm when I am certain.

puterboy, 2024-08-07:
Just a correction to my previous post - even the LTE device does have a 900MHz transmitter though it seems to transmit readings about every 2 hours This compares with up to about once per minute for standard packet R900 -- though the consumption data only changes at most once every 15 minutes (which is how often I believe it queries the register).

Also, it reports in 100ths of a cubic foot of water -- or just under 10oz which is not bad...

lnalsafe, 2024-11-25:
Flow meter, water flow meter, air flow meter, gas meter, gas leak detector, natural gas leak detector manufacturer.
WhatsApp: +8613437856962, Email: lnalsafe@gmail.com
https://www.lnalsafe.com/

closed 2025-08-15 by bemasher