bemasher/rtlamr #314: Error connecting to spectrum server

rstremke, 2024-11-11:
I just installed rtlamr on a brand new installation of Debian on a Raspberry Pi. Everything went well, but when trying to run any rtlamr command, I get the following:

main.go:90: Error connecting to spectrum server: dial tcp 127.0.0.1:1234: connect: connection refused rcvr.Connect
main.(*Receiver).NewReceiver 
/root/go/pkg/mod/github/.com/bemasher/rtlamr@v0.9.3/main.go:90
main.main
/root/go/pkg/mod/github/.com/bemasher/rtlamr@v0.9.3/main.go:313
runtime.main
/usr/local/go/src/runtime/proc.go:272
runtime.goexit
/usr/local/go/src/runtime/asm_arm.s:884

I've found a few other references to this type of error, but they seem to involve a Docker installation method, or a problem with a USB device becoming unavailable after time. I have not been able to successfully run this, yet.

Thanks in advance for any assistance.

Nils154, 2024-11-12:
Not at my computer, but pretty sure you need to start rtltcp firstI think the manual suggests running it in two separate shell windows--Nils Gokemeijer (Sent from iPhone)

main.main

rstremke, 2024-11-12:
I confirmed I have rtl_tcp running via a script. A systemctl status rtlsdr returns all green enabled/active (running) statuses. The call to rtlamr still returns the same error. By separate shell do you mean I should run rtl_tcp in zsh and rtlamr in bash - or something similar? Apologies, most of my unix experience is from about 20 years ago.

iSynthetixx, 2025-03-05:
That error for me happened when my docker container running rtlamr couldn't find my rtl_tcp server because the IP address changed. Try specifying the IP address of the machine where rtl_tcp is running explicitly rather than relying on the loopback address (127.0.0.1).